MatterHackers/MatterControl #1857: Not retracting on infill

galaxyman7, 2017-01-28:
On a certain model, retraction is not triggered when printing infill. This causes weak infill and weak perimeters because the nozzle oozes and a lot of the nozzle pressure is lost when moving from one infill line to the next. See the picture below. 
[image]

galaxyman7, 2017-01-28:
This is using MatterSlice by the way. This does not happen with Cura (standalone version).

larsbrubaker, 2017-01-31:
Can you attach or link me to the file? It is possible to debug this without it but it might be easier if I have your settings and STL.  Thanks.

galaxyman7, 2017-01-31:
Here's a link to the files:
https://drive.google.com/open?id=0B8kD8aYGny2acWp1WkZKZUJEY28
I used the "deckbox" presets for quality and "Green Excelvan" for material 1. 
Thanks

markleinhos, 2017-02-01:
Don't know if I'm seeing the same thing or a different thing, but I've noticed that when switching from infill to the perimeters of the next layer (basically next layer changes) it's not retracting. It's also not retracting for me when changing islands despite being checked on. Using MatterSlice.

galaxyman7, 2017-02-01:
Yeah I see a few issues here and there with it not retracting when crossing open space.

galaxyman7, 2017-02-03:
I just tried 1.6.2, and I get the same issue.

jlewin, 2017-02-06:
Fixed in master, should be in the next alphas

closed 2017-02-06 by jlewin