MatterHackers/MatterControl #1861: FR: Percent of print complete as a G-Code Variable

rytz, 2017-01-30:
As requested by a user, would like the ability to output the current build percentage as a G-Code variable so certain MakerBot models can display the build percentage when printing from SD card. 

http://forums.matterhackers.com/topic/380/problems-with-mattercontrol-and-ctc-bizer/12

http://wiki.mattercontrol.com/Custom_G-Code_Variables

unlimitedbacon, 2017-01-30:
Related issues:
https://github.com/MatterHackers/MatterControl/issues/1761
https://github.com/MatterHackers/MatterSlice/issues/176

stuartpb, 2017-12-24:
Heck, I'd settle for just having MatterControl directly send M73 commands based on the time-based percentage it displays in the main window.

larsbrubaker, 2018-09-24:
Closing - fixed. This has been added 

[image]

Please file a new issue if error persists.

closed 2018-09-24 by larsbrubaker