MatterHackers/MatterControl #1869: Where is the skirt command?

EngPhysGuy, 2017-02-01:
Just upgrading to 1.6 and I cannot find the skirt command. The replacement seems to be priming which is great but skirt followed the contours of my print to help prevent lift off.

Has skirt been removed or have I missed something?

unlimitedbacon, 2017-02-01:
The option you are looking for is called [First Layer Expansion](http://wiki.mattercontrol.com/SETTINGS/General/Raft-Priming/First_Layer_Expansion/Loops). It is also often called a brim. This creates attached loops around the base of the part to help hold it to the bed.

[Priming loops](http://wiki.mattercontrol.com/SETTINGS/General/Raft-Priming/Priming/Loops), also called a skirt, are used to get the plastic flowing through the hot end before it actually starts printing.

EngPhysGuy, 2017-02-01:
Hello

That solved my problem, thanks for the quick reply … ☺

Cheers… Peter

Peter Jonasson
Technical Supervisor, Engineering Physics
McMaster University
JHE-A304
1280 Main Street West
Hamilton, ON Canada
905-525-9140 x24935
peter.jonasson@mcmaster.ca<mailto:peter.jonasson@mcmaster.ca>
http://engphys.mcmaster.ca/contact-us/staff/lab-tech_2/

“Achieving the attainable makes the unattainable a possibility.”

closed 2017-02-01 by unlimitedbacon