MatterHackers/MatterControl #1924: Slic3r crashes when trying to slice anything on MC 1.6.2.

Fishbowlkraken, 2017-02-17:
This is on Ubuntu 16.04, dunno if anyone else is experiencing this. So, I'd finally upgraded to MatterControl 1.6.2 in the hopes of being able to use the current version of Slic3r and not print outside my build volume as it had been. However, I've found upon running it that Slic3r simply does not work. Every time I attempt to slice something it gives "slicing error, check settings." I wasn't sure if this was just due to my printer's custom settings, so I set up a generic Rostock Max profile, and lo and behold, it's still failing. Does this possibly have something to do with how the build volume is defined? I know previous versions of MC had issues with that. Until this is resolved I have to downgrade back to MC 1.4 again, so I'd really like to be able to resolve this.

larsbrubaker, 2017-02-21:
Tested and should be working on the latest alpha build (available now in Windows)

closed 2017-02-21 by larsbrubaker

Fishbowlkraken, 2017-02-22:
My point was this is an issue on Linux, so I don't see why you just closed
this.

Fishbowlkraken, 2017-02-22:
I suppose if you make a point release available on Linux for once I'd say
this was closed, but as this is still broken for Linux users, and my bug
report was concerning specifically the Linux version of your application, I
think you should keep this open until you release a fix for the platform
this ticket concerns.