MatterHackers/MatterControl #1929: Raspberry Pi Touchscreen scale factor

unlimitedbacon, 2017-02-18:
In touchscreen mode on the desktop MatterControl uses a UI scaling factor of 1.3. However, in order to fit everything on the Raspberry Pi display this needs to be reduced to 1.0 (the same as on Android, iirc).

Until now, I have been temporarily overriding this by commenting out line [263 in MatterControlApplication.cs](https://github.com/MatterHackers/MatterControl/blob/master/MatterControlApplication.cs#L263).

```
                        if (UserSettings.Instance.IsTouchScreen)
                        {
                                //GuiWidget.DeviceScale = 1.3;
                                SystemWindow.ShareSingleOsWindow = true;
                        }
```

Maybe we should have a way to configure this setting without changing the source code?

larsbrubaker, 2017-02-18:
Device scale is attempting to adjust the fact that the pixels are smaller and is set to 1.3 on touchscreen to help get bigger buttons on Android.

We can look at how to make this soft or configurable.

larsbrubaker, 2018-09-24:
Should be fixed in 2.0

closed 2018-09-24 by larsbrubaker