MatterHackers/MatterControl #1936: X3G is not an output option on Fedora

adammhaile, 2017-02-22:
I build MC from source on Fedora 25 using the instructions in the Readme verbatim. Both with the current master HEAD and with the Releases/1.6.2 tag and in both cases I don't get an X3G export option for my MakerBot Replicator 2 as per the instructions here: http://wiki.mattercontrol.com/Export_G-Code_as_X3G

Granted, I realize those are out of date. But everything I've seen says that the latest should have X3G built in automatically. And I can even see in the Replicator 2 profile (https://github.com/MatterHackers/MatterControl/blob/a74dfe56ffe72096bc06835e4d34e4ed8f1e133b/StaticData/Profiles/MakerBot/Replicator%202.printer#L166) that it's default output type is X3G.

Any thoughts on what I might be doing wrong?

unlimitedbacon, 2017-02-22:
Sorry but the X3G plugin is not currently open source. As far as I know, we plan to release it but just haven't gotten around to it yet. However, if you are running from source you can add it by copying in the .dll from the official release build. There are further instructions [on the wiki](http://wiki.mattercontrol.com/Development/Running_on_Linux#Adding_Proprietary_Plugins).

adammhaile, 2017-02-22:
That would explain it! I guess I assumed it was just using GPX or something. Thanks.
I'm probably going to switch the system in question to Ubuntu anyways, so no big deal for now... but would certainly love some official love for Fedora!

closed 2017-02-22 by adammhaile