MatterHackers/MatterControl #1960: Verson 1.6.2 Occasionally can't find hard drive.

Lloyd-H1, 2017-02-27:
This deals only with the newest version of MatterControl. Old version did not have this problem. New version of MatterControl shows an error that it can not find my hard drive after it starts up and is looking to load the 1st stl file. The error states that my hard drive is missing. This error shows up only the 1st time MatterControl tries to access my hard drive to load a new stl file, and the error continues even while accessing my hard drive in a different window. Hitting the "Continue" or "Try Again" simply repeats the error until an alternate window of my hard drive files is displayed. Hitting "Cancel" closes MatterControl entirely. But loading files afterwards does not show any errors.

Running Windows 7-64. 1 TB internal hard drive in a laptop, plus 2 external hard drives with a total of 3 TB in external storage., or 4 TB of storage total Recently upgraded to MatterControl 1.6.2. Wanhao i3 attached. Also attached is a docking port with large screen TV and Epson All-in-One printer-scanner connected.

unlimitedbacon, 2017-02-27:
This sounds more like a problem on the OS level than in MatterControl. Can you post a screenshot of the error message? Can you also try [deleting or renaming your application data](http://wiki.mattercontrol.com/Frequently_Asked_Questions#How_do_I_clear_the_application_data.3F)?

Lloyd-H1, 2017-02-28:
It is only happening with MatterControl, and it referred to harddrive5/DR Whatever that is.  I have the 3 hard drives, plus the card reader and the CD ROM. That's all that I have one this computer. I am uploading the screen shot. I have MatterControl reduced in this jpg so I could see some of my background programs. But the error is showing. (The 1st screen shot did not get saved to the clipboard for some reason, or else the clipboard got emptied before I could save it. This is the 2nd one.)

[image]

Lloyd-H1, 2017-02-28:
Here is a screenshot from today. With the newest version, it happens every time when I 1st try to access my hard drive to load a file. After the 1st time, there are no problems. And this happens only with the newest version of MatterControl. 
[image]

[image]

Oops. In trying to upload the newest screenshot just now, I am also getting an error message from the website here: "Something went really wrong, and we can’t process that file.  Choose Files"
Tried with 2 jpg's and a png file. Can't get it to upload. Will try from different web browser.

[image]
[image]

Lloyd-H1, 2017-02-28:
Let's try one more time with upload the screen shot. I was using Google Chrome browser before when my upload failed. Let's try it with Firefox.

[image]

Lloyd-H1, 2017-02-28:
Uploads to here are failing for some reason. Will try from a different computer.

Lloyd-H1, 2017-02-28:
One more time with today screenshot.
It appears that I can not upload the latest screen shot, even from my Linux computer. I may have accidentally closed this thread the last time. I will start a new thread.
The above problems have all been with my Windows 7 laptop, which is what I have hooked to my Wanhao i3.
I will start a new thread.

[image]

closed 2017-02-28 by Lloyd-H1