MatterHackers/MatterControl #1964: Newest version of MatterControl can't find drive. (New thread.)

Lloyd-H1, 2017-02-28:
I think that I accidentally closed the old thread, so I am starting a new one here.
I updated MatterControl on my Windows 7-64 laptop. It has a total of 4 TB of hard space connected to it using 2 external drives. This is also the computer which I have hooked into my Wanhao i3 3D printer.

The new version of Matter Control shows me an error of being unable to find a hard drive. The image will be below (I hope). I hit the button "Try Again" repeatedly and it will eventually find the drive. This problem ONLY occurs when using the new version of MatterControl. It did not happen with the older version. And this only happens the 1st time that I try to add a file after opening MatterControl. Loading files later does not show the error.

The error shows "There is no disk in the drive. Please insert a disk into drive \Device\Harddisk5\DR5"

I have no idea what Harddisk5\DR5 is referring to. 

If I close Matter Control, then re-open it, the program will again show the error the 1st time that I try to load a file. But I don't get any errors afterwards while MatterControl is open and active.

Screenshot.
Can't seem to upload screenshot image here today from any computers. I tried from my Windows 7-64 computer, and now from my Linux Mint computer. I have a high-speed connection, but I get an error from my Windows computer, and my Linux Mint computer simply sits here without uploading the image.

I finally got the same error on my Linux computer that I got on my Windows computer when I tried to upload the screenshot: "Something went really wrong, and we can’t process that file. "**_

Tried to upload jpg and png versions of it.

jlewin, 2017-03-01:
Original Issue with screenshots - MatterHackers/MatterControl#1960
[image]

jlewin, 2017-03-01:
Looks very similar to this bug report:
https://connect.microsoft.com/VisualStudio/feedback/details/615558/no-disk-pop-up

As far as I know we're not shipping symbols though. Will double back tomorrow and make sure no .pdb files are bundled in with release builds.

Lloyd-H1, 2017-03-01:
Yes, it does seem to be the same bug. My cure is to hit "Retry" enough times so that the Documents folder will finally come up and is displayed. It only happens the on the 1st time that I try to add a stl file to MatterControl. Afterwards, everything always works fine until I close MatterControl. Then upon re-opening MatterControl and adding the 1st stl file again, I will again get this same error. I don't get this error with any other program.

I have MatterControl on my Linux Mint computer also. I do not have this problem with the Linux version. But I only use my Windows 7 computer with the 3D printer. (I simply use MattelrControl on the Linux computer to checkout files. The Linux computer is my main computer for everything else.)