MatterHackers/MatterControl #1976: When the printer is on MatterControl keeps the heated bed on

kailledargun, 2017-03-05:
Using a robo r1+ when ever I try to turn the heated bed off it stays on. The only way to get it to cool down is to shut the printer off completely. Did not happen until I updated to newest software.

larsbrubaker, 2017-03-10:
Can you let me know what build number you are on? We are not seeing this but I want to verify on the same build as you in case it was a regression. 

Have you checked this on the latest Alpha? I would love to know if it is still happening.

Thanks.

kailledargun, 2017-03-12:
I reverted back a couple of versions as I needed some prints complete. I will be away from the printer but as soon as I can I will update again and see what happens

kailledargun, 2017-04-03:
I have tried again on 1.6 on Ubuntu 16. I also tried 1.7 on Windows 10. I looked at the gcode that is being produced and I do not see a command to shut down the bed warmer. The extruder shuts off but not the bed.