MatterHackers/MatterControl #1977: MatterControl does not run on Linux with Nvidia driver version 378.13

unlimitedbacon, 2017-03-06:
See the [report on Reddit](https://www.reddit.com/r/archlinux/comments/5xnkad/mattercontrol_nvidia_378/).

The app doesn't seem to crash since there is no exception or stack trace, however the UI never appears. Tested on Arch with a GeForce GTX 1070, 970M, and 670. Nvidia driver version 378.13.

h54, 2017-03-10:
I have the same issue on Ubuntu 16.04 with a GTX-970M

didli, 2017-03-23:
Same here in Fedora 25 x64 with a Geforce GTX 750 Ti.
I think the problem was beginning to show up even before the 378.13 nvidia driver release, and after the latest updates from mono-project.

wahoo8, 2021-05-25:
Just edit the color setting in /usr/lib/mattercontrol appsettings.json to be 24 instead of 32.  This will work just fine after that.