MatterHackers/MatterControl #1982: Matterslice is ignoring top layer setting

PlatnumK, 2017-03-09:
It seems that regardless of top layer setting Matterslice only prints 1 top layer at current layer hight setting.

For example if my layer hight is 0.2mm and my number of top layers are set to 3,  And then if I slice a project using Matterslice, When it come to doing the top layers it will only do 1 layer at 0.2mm instead of 3 layers.  Yet if I was to slice the same project with Cura instead it will do all 3 top layers.

larsbrubaker, 2017-03-10:
Thanks for the info, definitely haven't seen this happening. 

Can you please attach your settings and stl to this issue? Also, what build are you on?

Again, thanks for the feedback.

PlatnumK, 2017-03-11:
I believe that I have found the cause. What I thought was happening and what was actually happening are 2 different things. 

After printing quite a few Calibration Cubes (and having 2 nozzle jams) I found that if I decreased the top infill speed, the problem seems to go away.

closed 2017-03-11 by PlatnumK