MatterHackers/MatterControl #1989: MatterControl and Ubuntu 16.10

lslewis901, 2017-03-13:
Trying to run the latest Linux version of MatterControl where the UI loads then stops and I get the following error stack in the console (when running from term using sudo):
So far all I was able to do was perform the initial login...

Web exception occurred. Status code: NameResolutionFailure
Web exception occurred. Status code: NameResolutionFailure
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Waiting for a connection on port 52742...

Unhandled Exception:
System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'System.Net.Sockets.Socket'.
  at System.Net.Sockets.Socket.ThrowIfDisposedAndClosed () [0x00021] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.Socket.GetSocketOption (System.Net.Sockets.SocketOptionLevel optionLevel, System.Net.Sockets.SocketOptionName optionName) [0x00000] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at SuperSocket.ClientEngine.TcpClientSession.ProcessConnect (System.Net.Sockets.Socket socket, System.Object state, System.Net.Sockets.SocketAsyncEventArgs e) [0x00060] in <4ba236fbf29248a4985a92e6e3907e4a>:0 
  at SuperSocket.ClientEngine.AsyncTcpSession.SocketEventArgsCompleted (System.Object sender, System.Net.Sockets.SocketAsyncEventArgs e) [0x0000c] in <4ba236fbf29248a4985a92e6e3907e4a>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.OnCompleted (System.Net.Sockets.SocketAsyncEventArgs e) [0x00014] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.FinishWrapperConnectSuccess (System.Net.Sockets.Socket connectSocket, System.Int32 bytesTransferred, System.Net.Sockets.SocketFlags flags) [0x00016] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.MultipleConnectAsync.Succeed () [0x0002d] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.MultipleConnectAsync.InternalConnectCallback (System.Object sender, System.Net.Sockets.SocketAsyncEventArgs args) [0x000d4] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.OnCompleted (System.Net.Sockets.SocketAsyncEventArgs e) [0x00014] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.Complete () [0x00000] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.Socket.<ConnectAsyncCallback>m__3 (System.IAsyncResult ares) [0x00062] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.SocketAsyncResult+<Complete>c__AnonStorey0.<>m__0 (System.Object _) [0x00000] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00019] in <f712f98eb8e445c8918edaf595bbe465>:0 
  at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00096] in <f712f98eb8e445c8918edaf595bbe465>:0 
  at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <f712f98eb8e445c8918edaf595bbe465>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'System.Net.Sockets.Socket'.
  at System.Net.Sockets.Socket.ThrowIfDisposedAndClosed () [0x00021] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.Socket.GetSocketOption (System.Net.Sockets.SocketOptionLevel optionLevel, System.Net.Sockets.SocketOptionName optionName) [0x00000] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at SuperSocket.ClientEngine.TcpClientSession.ProcessConnect (System.Net.Sockets.Socket socket, System.Object state, System.Net.Sockets.SocketAsyncEventArgs e) [0x00060] in <4ba236fbf29248a4985a92e6e3907e4a>:0 
  at SuperSocket.ClientEngine.AsyncTcpSession.SocketEventArgsCompleted (System.Object sender, System.Net.Sockets.SocketAsyncEventArgs e) [0x0000c] in <4ba236fbf29248a4985a92e6e3907e4a>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.OnCompleted (System.Net.Sockets.SocketAsyncEventArgs e) [0x00014] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.FinishWrapperConnectSuccess (System.Net.Sockets.Socket connectSocket, System.Int32 bytesTransferred, System.Net.Sockets.SocketFlags flags) [0x00016] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.MultipleConnectAsync.Succeed () [0x0002d] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.MultipleConnectAsync.InternalConnectCallback (System.Object sender, System.Net.Sockets.SocketAsyncEventArgs args) [0x000d4] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.OnCompleted (System.Net.Sockets.SocketAsyncEventArgs e) [0x00014] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.Complete () [0x00000] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.Socket.<ConnectAsyncCallback>m__3 (System.IAsyncResult ares) [0x00062] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Net.Sockets.SocketAsyncResult+<Complete>c__AnonStorey0.<>m__0 (System.Object _) [0x00000] in <5641e4edad4f4464ba58c620a7b8ea48>:0 
  at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00019] in <f712f98eb8e445c8918edaf595bbe465>:0 
  at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00096] in <f712f98eb8e445c8918edaf595bbe465>:0 
  at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <f712f98eb8e445c8918edaf595bbe465>:0

unlimitedbacon, 2017-03-13:
Another user reported this issue via Help Scout (ticket #12515).

I did some investigation and am seeing the same issue with the current release and alpha builds for Linux, but not when building from source. I suspect it may have something to do with the recent release of Mono 4.8. In any case it looks like the only think we have to do is put out a new release.

unlimitedbacon, 2017-03-16:
Correction. This is now happening when building from source as well. @jlewin, I have a consistent repro on my computer. Can you take a look at it when you get a chance?

jlewin, 2017-03-16:
It looks like WebSocket4Net isn't compatible with Mono 4.8.0. Unlike in 4.6.x where the SocketIO library simply wouldn't connect, it now blows up with the System.ObjectDisposedException in SuperSocket.ClientEngine.TcpClientSession.ProcessConnect()

As before I think we remain pinned on 4.4.2 for the time being:
http://wiki.mattercontrol.com/Development/Running_on_Linux#Installing_Mono

unlimitedbacon, 2017-03-17:
Should we file a bug upstream with WebSocket4Net?

jlewin, 2017-03-17:
Definitely, are you offering or would you like me to?

unlimitedbacon, 2017-03-17:
I wouldn't know what to tell them. Could you do it?

jlewin, 2017-03-17:
Will do

unlimitedbacon, 2017-03-17:
I tried updating WebSocket4Net to version 0.15.0-beta6 and so far it is working better. The stable version we have been using (0.14.1) is quite old.

jlewin, 2017-03-28:
Let's try adding a WebSocket4Net nuget dependency at the beta version and make builds to confirm

didli, 2017-04-21:
@unlimitedbacon Could you share how to update WebSocket4Net ? I'm stuck with this issue for two months now. 
And I'm not seeing much activity from kerryjiang/WebSocket4Net.

unlimitedbacon, 2017-10-12:
This should be fixed now. Can you guys please try the latest alpha?
[MatterControl Linux Alpha - 1.7.5.9469](https://mattercontrol.appspot.com/downloads/development/ag9zfm1hdHRlcmNvbnRyb2xyQQsSB1Byb2plY3QYgICAiOCSzAsMCxINUHVibGljUmVsZWFzZRiAgIDg8dSdCgwLEgZVcGxvYWQYgICA4NvAngkM)

lslewis901, 2017-10-12:
Ok, Seems to work fine now.  Did have a hang issue at one point during loading of 3D object sample file for test print.  Restarted the app and it was fine.

lslewis901, 2017-10-12:
Here is actual stack trace from syslog (with main error as Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]: 84#011../sysdeps/unix/syscall-template.S: No such file or directory.)

[Thread debugging using libthread_db enabled]
Oct 12 11:40:06 lindsey-ThinkPad-T500 compiz[2161]: Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]: 0x00007f727b53bd8d in poll () at ../sysdeps/unix/syscall-template.S:84
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]: 84#011../sysdeps/unix/syscall-template.S: No such file or directory.
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   Id   Target Id         Frame
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]: * 1    Thread 0x7f727c547740 (LWP 32531) "Main" 0x00007f727b53bd8d in poll () at ../sysdeps/unix/syscall-template.S:84
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   2    Thread 0x7f727abff700 (LWP 32533) "SGen worker" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   3    Thread 0x7f7278834700 (LWP 32535) "Finalizer" 0x00007f727ba2e08a in __waitpid (pid=2681, stat_loc=0x7f727883286c, options=0) at ../sysdeps/unix/sysv/linux/waitpid.c:29
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   4    Thread 0x7f727314f700 (LWP 32536) "Main" pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   5    Thread 0x7f727310a700 (LWP 32537) "Threadpool work" pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   6    Thread 0x7f725be5a700 (LWP 32539) "dconf worker" 0x00007f727b53bd8d in poll () at ../sysdeps/unix/syscall-template.S:84
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   7    Thread 0x7f725b659700 (LWP 32540) "gmain" 0x00007f727b53bd8d in poll () at ../sysdeps/unix/syscall-template.S:84
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   8    Thread 0x7f725ae58700 (LWP 32541) "gdbus" 0x00007f727b53bd8d in poll () at ../sysdeps/unix/syscall-template.S:84
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   9    Thread 0x7f72582c1700 (LWP 32542) "Timer-Scheduler" pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   10   Thread 0x7f72429ba700 (LWP 32543) "Threadpool work" 0x00007f7278a5c5cc in Mono_Globalization_Unicode_MSCompatUnicodeTable_Category_int (cp=112) from /usr/lib/mono/aot-cache/amd64/mscorlib.dll.so
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   11   Thread 0x7f7258d7b700 (LWP 32544) "Threadpool work" 0x00007f727b53bd8d in poll () at ../sysdeps/unix/syscall-template.S:84
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   12   Thread 0x7f724257b700 (LWP 32548) "Threadpool work" pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   13   Thread 0x7f723bd7a700 (LWP 32550) "LocalListener" 0x00007f727ba2d93d in accept () at ../sysdeps/unix/syscall-template.S:84
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   14   Thread 0x7f723bb79700 (LWP 32551) "Threadpool work" pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   15   Thread 0x7f723b64b700 (LWP 32552) "Threadpool work" 0x0000559c7af54a4b in ?? ()
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   16   Thread 0x7f723a16a700 (LWP 32558) "Threadpool work" 0x0000000040df303f in ?? ()
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   17   Thread 0x7f7239f69700 (LWP 32560) "Threadpool work" 0x0000000040e0fd26 in ?? ()
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   18   Thread 0x7f7239d68700 (LWP 32561) "Threadpool work" pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   19   Thread 0x7f723bf7b700 (LWP 2607) "Threadpool work" pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   20   Thread 0x7f723b249700 (LWP 2616) "Threadpool work" pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]:   21   Thread 0x7f720b194700 (LWP 2678) "pool" syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
Oct 12 11:40:07 lindsey-ThinkPad-T500 compiz[2161]: Thread 21 (Thread 0x7f720b194700 (LWP 2678)):

unlimitedbacon, 2017-10-12:
Looks like this is all Compiz related stuff, not MatterControl. There is no Mono stack trace there. As far as I know Mono does not report errors through syslog, only the terminal. If it happens again, check the terminal.  If the stack trace doesn't mention `System.Net.Sockets.Socket` then its a different bug and warrants opening separate bug report.

lslewis901, 2017-10-12:
Ok, sounds good. I think the app sent an error report anyway. And I did get
it to work on restart.
Thanks

lslewis901, 2017-10-12:
Ok, Maybe not related to original issue, but the app does crash everytime I load a new file,  However the file is there when I restart MatterControl.