MatterHackers/MatterControl #2000: Temperatures not registering. Mac OSX 10.12.3  Matter Control 1.6 / 1.7.0

ArtDezign, 2017-03-20:
I am able to communicate with my printer in every way except for getting the hot end, and bed temperatures.  I am able to send signals to heat these elements up, and can read the temperatures being provided to MatterControl via the terminal, but they are not connected to the "actual" temperature provided in the GUI.  This means I cannot print because matter control thinks my thermistors are at 3deg, and 0deg respectively. 

I issued this bug with tech support and they had me wait for beta 1.7.0 claiming it was going to be fixed, but it doesn't appear as though it is.  I looked for the same issues here, but didn't see it, so sorry if this is a duplication.

unlimitedbacon, 2017-03-21:
What kind of printer do you have? What firmware does it run? Can you send an `M105` through the terminal and then post a [terminal log](http://wiki.mattercontrol.com/Send_Debugging_Information#Print_Log_.2F_Terminal_Log)?

ArtDezign, 2017-03-21:
I have a Geeetech Rostock 301 printer.  The firmware is Rostock301_V1.0.02, on a GTM32 control board.

I'm attaching two terminal logs.  The one labeled with M105 is from v.1.7.0.  The other file is the same issue but running from v.1.6. This log contains higher temperatures.

Let me know if there is anything else I can do to help.

[Rostock301_M105_terminal Log.txt](https://github.com/MatterHackers/MatterControl/files/856827/Rostock301_M105_terminal.Log.txt)
[Rostock301_Matter_control_terminal_Log.txt](https://github.com/MatterHackers/MatterControl/files/856828/Rostock301_Matter_control_terminal_Log.txt)

larsbrubaker, 2017-03-21:
After investigating the attached files (thank you), it has to do with the parsing of the temperature response. The firmware for your printer is outputting is different that what we have support until now.

marlin firmware starts with "ok T0:"
smoothie starts with "B:"
repetier contains a "T:" somewhere in the response

This firmware has a T0 but not at the start of the line.

We'll fix it :).  

Thanks again.

ArtDezign, 2017-04-16:
So, is there an ETA on when the next beta is released? I currently cannot use matter control for my printer because of this bug, so if it's been solved I'd love to get that update.