MatterHackers/MatterControl #2007: Line numbering issues slowing down Mattercontrol 1.7 with Marlin Bugfix

Dreamloom, 2017-03-29:
Is there anyway to keep MC 1.7 from demanding line number data or whatever it is doing now - that it was not doing in version 1.6.2? It is slowing USB communication with the Arduino to the point where the motors are stuttering when they are normally very smooth.

larsbrubaker, 2017-04-03:
Those are the checksum values that marlin uses to ensure a line is sent correctly. In 1.7 we now send all lines with checksums (even when not printing) to ensue they are received correctly.

The behavior while printing should not have changed (it always has sent checksums while printing). Can you verify that your are seeing a difference while printing?

Is it possible that there is something else running on that machine that was not running before?

Thanks for the feedback we'll try to investigate other possible causes.

duisenberg, 2017-07-19:
I would apply too for a checkbox in printer settings to disable checksums, since I also have "stuttering" issues, especially on tiny lacy parts of the print and when printing circles.  My suspect is Marlin is sometimes a bit behind with planner, checksum and menu polling.

duisenberg, 2017-09-20:
Setting serial speed to 250kbps and TX_BUFFER_SIZE in Marlins Configuration_adv.h to 256 solved the "stuttering" issue for me. (Has been 115k2 and 0 before)

duisenberg, 2017-10-05:
I must correct that - the stuttering issue was solved for some models, but not at all. 

It might depend a bit on the models tesselation, but for some arcs it slows down and paces up again and again with no obvious reason. Thought M105 could be polled a little less, but that may be coincidence 
when watching the command stream in terminal.