MatterHackers/MatterControl #2011: Linux version 1.7.0.8527 - no file dialog when clicking +ADD

BigD3m0n, 2017-04-01:
I just upgraded to Linux version 1.7.  When I click ADD to select a new file no file dialog shows up.  I get no error messages either.  I can drag and drop a file into the queue but clicking ADD will not work for me.  Verison 1.6 worked without issue for me and the install of 1.7 went through without error.  

DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=17.2
DISTRIB_CODENAME=rafaela
DISTRIB_DESCRIPTION="Linux Mint 17.2 Rafaela"
NAME="Ubuntu"
VERSION="14.04.2 LTS, Trusty Tahr"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 14.04.2 LTS"
VERSION_ID="14.04"

unlimitedbacon, 2017-04-04:
Sometimes it shows up behind the main window. Can you verify that the dialog box is not appearing behind MatterControl?

If not, can you try installing the gtk-sharp2 package?

BigD3m0n, 2017-04-04:
No window anywhere that I can see.  So I installed gtk-sharp2.  Now when I click +ADD MatterControl crashes and I get this error on the commandline.  I then attempted to run MatterControl as root and it does bring up the file dialog but it is huge.
[image]

(process:28199): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(process:28199): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_size_lookup_for_settings: assertion 'GTK_IS_SETTINGS (settings)' failed

(process:28199): Gtk-WARNING **: Invalid icon size 1

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_theme_load_icon: assertion 'GTK_IS_ICON_THEME (icon_theme)' failed

(process:28199): Gtk-WARNING **: Error loading theme icon 'edit-find' for stock: 

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_size_lookup_for_settings: assertion 'GTK_IS_SETTINGS (settings)' failed

(process:28199): Gtk-WARNING **: /build/gtk+2.0-KsZSEA/gtk+2.0-2.24.23/gtk/gtkstyle.c:2358: invalid icon size '1'

(process:28199): Gtk-CRITICAL **: IA__gtk_style_render_icon: assertion 'pixbuf != NULL' failed

(process:28199): GLib-GObject-CRITICAL **: g_object_ref: assertion 'G_IS_OBJECT (object)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_theme_load_icon: assertion 'GTK_IS_ICON_THEME (icon_theme)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_size_lookup_for_settings: assertion 'GTK_IS_SETTINGS (settings)' failed

(process:28199): Gtk-WARNING **: Invalid icon size 1

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_theme_load_icon: assertion 'GTK_IS_ICON_THEME (icon_theme)' failed

(process:28199): Gtk-WARNING **: Error loading theme icon 'text-x-generic' for stock: 

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_size_lookup_for_settings: assertion 'GTK_IS_SETTINGS (settings)' failed

(process:28199): Gtk-WARNING **: /build/gtk+2.0-KsZSEA/gtk+2.0-2.24.23/gtk/gtkstyle.c:2358: invalid icon size '1'

(process:28199): Gtk-CRITICAL **: IA__gtk_style_render_icon: assertion 'pixbuf != NULL' failed

(process:28199): GLib-GObject-CRITICAL **: g_object_ref: assertion 'G_IS_OBJECT (object)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_theme_lookup_by_gicon: assertion 'GTK_IS_ICON_THEME (icon_theme)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_theme_lookup_by_gicon: assertion 'GTK_IS_ICON_THEME (icon_theme)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_icon_theme_lookup_by_gicon: assertion 'GTK_IS_ICON_THEME (icon_theme)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:28199): GLib-GObject-CRITICAL **: g_object_get: assertion 'G_IS_OBJECT (object)' failed

(process:28199): Gtk-CRITICAL **: IA__gtk_settings_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed

(process:28199): GLib-GObject-CRITICAL **: g_object_get: assertion 'G_IS_OBJECT (object)' failed
[... 134 lines cut ...]

(process:28199): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

(MatterControl:28199): Gtk-WARNING **: Screen for GtkWindow not set; you must always set
a screen for a GtkWindow before using the window

(MatterControl:28199): Gdk-WARNING **: Native Windows wider or taller than 32767 pixels are not supported

(MatterControl:28199): Gtk-WARNING **: Screen for GtkWindow not set; you must always set
a screen for a GtkWindow before using the window

(MatterControl:28199): Gtk-WARNING **: Screen for GtkWindow not set; you must always set
a screen for a GtkWindow before using the window

(MatterControl:28199): Gtk-WARNING **: Screen for GtkWindow not set; you must always set
a screen for a GtkWindow before using the window
The program 'MatterControl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'RenderBadPicture (invalid Picture parameter)'.
  (Details: serial 915 error_code 143 request_code 140 minor_code 8)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

unlimitedbacon, 2017-04-04:
I am unable to reproduce your error. I installed a fresh copy of Linux Mint 17.2 MATE in a VM. After installing gtk-sharp2, the dialog boxes work correctly. Is there anything else unusual about your environment? The errors are the sort of thing you run into when the `DISPLAY` environment variable is not set, which would be the case if you were launching MatterControl from outside of X (like over ssh). Also, what version of Mono do you have (run `mono --version`)?

@gregory-diaz, can you add gtk-sharp2 as a dependancy to the Debian package and make a new alpha build?

BigD3m0n, 2017-04-04:
I am running Linux Mint 17.2 KDE.

This is what I get for mono --version:
Mono JIT compiler version 4.4.2 (Stable 4.4.2.11/f72fe45 Fri Jul 29 09:58:49 UTC 2016)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
        TLS:           __thread
        SIGSEGV:       altstack
        Notifications: epoll
        Architecture:  amd64
        Disabled:      none
        Misc:          softdebug 
        LLVM:          supported, not enabled.
        GC:            sgen

BigD3m0n, 2017-04-05:
Also, if I revert back to build 1.6.1.8140, linux stable, I have no issues.

closed 2017-04-29 by larsbrubaker