MatterHackers/MatterControl #2066: Azteeg X5 Mini with Smoothie stutters

rytz, 2017-05-16:
User reports via email that the Azteeg X5 Mini and Smoothie firmware produces stuttering moves. When moves are broken down into smaller moves, each smaller move is made independent from the others, so the overall action is not smooth, but jerky.

From the user: 

It's a cartesian printer, a diy clone TEVO Tarantula I've had for over a year (heavily modded but the same basic setup, single Z).
I'm running the latest edge firmware dated 5-10-17 from smoothies github.

If it matters, some more specs:
24V 350W meanwell psu
X5 mini V3
SD6128 digipot drivers from Panucatt
Mini VIKI display w/single ribbon cable
Communication is through a gigabit switch, not a router.
The bed is powered through an external MOSFET from a 12V Seasonic ATX PSU.

I did just add a PT-100 that needed a large section of the hotend temperature config pasted in, otherwise it's largely unmodified, and also fresh from github lastnight under the Edge>softendstops branch (really didn't see anything different than Arthur's draft though), and it is specifically for the X5 mini V3.
The PT-100 is on 3v3, GND & pin 1.30 and it's working fine (if a little high at room temp).

whosawhatsis, 2017-05-16:
I just saw the same issue. It's a communication problem. The buffer is running out because moves are being executed faster than they are being transmitted. The solution is to run the gcode from the SD card, not over USB.

Diggrr, 2017-05-16:
Connected via ethernet through a gigabit switch, brand new sdhc card too.

Diggrr, 2017-05-16:
Sorry, this is my machine with the issue..
Homing moves smooth as silk, it is all other moves, whether by matter control manually or during a live print. Pronterface has no issue that I've seen.
All MC movements (10-100, + or -) are broken into 2.5mm segments for X&Y and 1mm segments for Z, and played individually one at a time with second or two between each small move.
Thanks for looking!

jlewin, 2017-05-31:
We'll add the option to disable line chopping in an upcoming build

larsbrubaker, 2019-02-05:
This has been done in 2.0

closed 2019-02-05 by larsbrubaker