MatterHackers/MatterControl #2096: make the g-code editor boxes adjustable/copy paste

mitchy2233, 2017-05-30:
it sure is a pain in the ass to write custom g-code!! i can't see half the stuff i put in as the box is so tiny and unadjustable!!why cant i just copy and paste my custom g-code why is this feature not included/available ??  why wouldn't you make it adjustable?? lol unless im missing something!! i honestly avoid using this slicer for this exact reason!! please fix this minor/easy issue!  until then i guess ill use a different slicer!! which sucks because i like this one!! 
please excuse me if there is a way to copy and paste or adjust the letter box!! if there is please inform me on how to do this!! you can email me the solution @ mitchhirsch@icloud.com

jlewin, 2017-05-31:
Copy paste currently works with the traditional keyboard keys (Ctrl-C and Ctrl-V) on the focused text field. What platform are you using and has it ever worked and/or did it recently stop working?

larsbrubaker, 2017-12-03:
This has been fixed and will be in the next release.

closed 2017-12-03 by larsbrubaker