MatterHackers/MatterControl #2138: Freakin out with retractions

duisenberg, 2017-06-07:
As for 1.7.1.8936 Mattercontrol goes mad with retractions:

[image]

Guess this his is not as retractions were meant.

STL and printer settings attached.

[Halter I.zip](https://github.com/MatterHackers/MatterControl/files/1058948/Halter.I.zip)
[Anycubic.zip](https://github.com/MatterHackers/MatterControl/files/1058952/Anycubic.zip)

duisenberg, 2017-06-07:
Been a little upset, sorry. Resetting retraction parameters to default 8 / 0.1 mm reduced retraction picture back to normal. 

Extrusion length alone seems to have no effect.

larsbrubaker, 2017-06-08:
Sorry about the confusion in the new build. Minimum Travel was not ALWAYS being respected, so fixing that bug likely introduced what you are seeing.

What are you seeing regarding 'Extrusion length alone seems to have no effect'? Both these variables should be respected. Very interested if you are finding a case where they are not.

[image]

Thanks. Please let me know if we need to take some action here, or close if not.

duisenberg, 2017-06-09:
Both variables are respected now. Have been a bit surprised from the sheer number of all those retract points - the vast majority doesnt make much sense for printing as with minimum settings retraction even happens where the tool path is closed, just changing direction. Usually retraction should only happen in travel moves - but there it should happen reliable. Which still is not the case in current version. 
But guess as issue this would be more related to Matterslice.

closed 2017-06-09 by duisenberg