MatterHackers/MatterControl #2146: iMe firmware does not support M400 command

rytz, 2017-06-09:
User reports having installed [iMe firmware](**https://github.com/donovan6000/iMe**) and an incompatibility with command M400:

`->N66 M400*23 [135.856]
<-Error: Unknown G-code command [135.858]`

Suggest an option to not send M400 if desired.

larsbrubaker, 2017-06-12:
M400 (wait for moves to finish) is very important to many of our commands. Firmware that does not support it is likely to be a problem for things like our advanced macros. What is the benefits of this firmware? Maybe we can look into supporting it.

rytz, 2017-06-12:
It's third-party firmware for the M3D to enable it to be used with G-Code-based slicers: https://github.com/donovan6000/iMe

Perhaps we can reach out to firmware creator donovan6000 and see about fixing the M400 on the firmware side.

jlewin, 2017-06-16:
Fixed

closed 2017-06-16 by jlewin