MatterHackers/MatterControl #2166: Print leveling stopped working

daeniz, 2017-06-14:
After updating via the stable channel, print levelling no longer works as intended. 

When configuring it homes as usual, setting z to 0.00 mm. 

In step two it travels to first probe point and sets Z to the starting height (10 mm in my case). When pushing -Z it fails, saying it extruder cannot go any lower (which is kinda odd, since it is at 10 mm, and was just at 0).

[image]

unlimitedbacon, 2017-06-14:
Someone else on the forum is reporting the same issue.
http://forums.matterhackers.com/topic/609/manual-bed-leveling-wizard-says-warning-moving-too-low

We've been able to replicate it in house as well. Currently the workaround is to go to Settings > Features > Behavior and check [Z Can be Negative](http://wiki.mattercontrol.com/SETTINGS/Printer/Features/Behavior/Z_Can_Be_Negative).

daeniz, 2017-06-15:
Work-around with Z can be negative works so far. Thank you for you quick reply and more than excellent service for an excellent and free-to-use product. You rock!

closed 2017-06-23 by larsbrubaker