MatterHackers/MatterControl #2202: Retraction sometimes does not occur

paziman, 2017-06-25:
Hi,

I'm using k8200 and I have been tetin MatterControl for some time. I for some reason retraction does not always occur and print produces somewhat similar like in the first picture in here http://www.matterhackers.com/news/how-to-use-support-material-part-1

By looking the printing it seems that before moving to another island it retracts and excratcs filament back before moving. I took consolelog open but for me its a bit mess to read.

If I'm correct here is a clip from console log where it can be seen that retraction doe not occur  - i have 5 mm retraction in my setup and when changing islands too.

`->N53526 G1 X109.59 Y123.34 E2785.009 F1080*56 [4347.427]

<-ok [4347.459]

->N53527 G1 X109.53 F2700*37 [4347.459]

<-ok [4347.475]

->N53528 G1 X104.58 E2785.126 F1080*86 [4347.475]

<-ok [4347.507]

->N53529 G1 X99.63 E2785.244*2 [4347.507]

<-ok [4347.523]

->N53530 G1 X94.69 E2785.362*8 [4347.523]

<-ok [4347.555]

->N53531 G1 X89.74 E2785.48*50 [4347.555]

<-ok [4347.763]

->N53532 G1 X89.68 F2700*16 [4347.763]

<-ok [4347.955]

->N53533 G1 X89.74 Y123.45 E2785.483 F1080*10 [4347.955]

<-ok [4348.146]

->N53534 G1 E2781.483 F3900*56 [4348.146]

<-ok [4348.162]

->N53535 G1 X90.4 Y123.84 F2700*78 [4348.162]

<-ok [4348.194]

->N53536 G1 X91.98 Y124.59*29 [4348.194]

<-ok [4348.226]

->N53537 G1 E2785.583 F3900*62 [4348.226]

<-ok [4348.274]

->N53538 M105*47 [4348.274]

<-ok T:189.5 /190.0 B:54.8 /55.0 @:127 [4348.322]

->N53539 G1 X92.06 Y124.6 E2785.584 F1200*51 [4348.322]

<-ok [4348.338]

->N53540 G1 X91.99 F2700*18 [4348.338]

<-ok [4348.354]

->N53541 G1 X92.23 Y124.76 E2785.589 F1200*1 [4348.354]

<-ok [4348.370]

->N53542 G1 Y124.8 F2700*22 [4348.370]

<-ok [4348.386]

->N53543 G1 X91.99 Y124.76 E2785.594 F1200*13 [4348.386]

<-ok [4348.402]

->N53544 G1 Z19.94 F2700*25 [4348.402]

<-ok [4348.418]

->N53545 G1 X90.4 Y123.84*42 [4348.418]

<-ok [4348.450]

->N53546 G1 X89.35 Y123.09*22 [4348.450]

<-ok [4348.482]

->N53547 G1 X87.91 Y123.48*18 [4348.482]

<-ok [4348.498]

->N53548 G1 X87.4 Y121.59*35 [4348.498]

<-ok [4348.514]

->N53549 G1 Y116.87*72 [4348.514]

<-ok [4348.786]

->N53550 G1 Y112.14*78 [4348.786]

<-ok [4349.057]

->N53551 G1 Y107.42*72 [4349.057]

<-ok [4349.329]

->N53552 M105*35 [4349.329]

<-ok T:188.9 /190.0 B:54.8 /55.0 @:127 [4349.617]

->N53553 G1 Y102.69*70 [4349.617]

<-ok [4349.633]

->N53554 G1 Y97.97*125 [4349.633]

<-ok [4349.649]

->N53555 G1 Y93.25*113 [4349.649]

<-ok [4349.729]

->N53556 G1 Y88.52*120 [4349.729]

<-ok [4349.761]

->N53557 G1 Y83.8*77 [4349.761]

<-ok [4349.824]

->N53558 G1 Y79.07*120 [4349.824]

<-ok [4349.936]

->N53559 G1 Y74.35*117 [4349.936]

<-ok [4349.952]

->N53560 G1 X87.91 Y72.46*44 [4349.952]

<-ok [4349.968]

->N53561 G1 X89.35 Y72.85*34 [4349.968]

<-ok [4350.080]

->N53562 G1 X89.68 Y72.6*18 [4350.080]

<-ok [4350.128]

->N53563 G1 X90.4 Y72.1*38 [4350.128]

<-ok [4350.176]

->N53564 G1 X92.6 Y70.45*19 [4350.176]

<-ok [4350.224]

->N53565 G1 X93.25 Y70.2 E2785.611 F1200*1 [4350.224]

<-ok [4350.272]

->N53566 G1 X93.72 Y70.09 E2785.622*94 [4350.272]`

I have a short mp4 video taken from this but even in a zip file github gives

"We don’t support that file type. with a PNG, GIF, JPG, DOCX, PPTX, XLSX, TXT, PDF, or ZIP. "

paziman, 2017-06-25:
Picture submitted but no video =/
[image]

larsbrubaker, 2017-06-26:
Thank you. We have another bug that is tracking this issue. I'll link this issue to that one and close this one.

MatterHackers/MatterControl#2203
Ignore minimum Extrusion on retraction when changing Islands

closed 2017-06-26 by larsbrubaker