MatterHackers/MatterControl #2229: Firmware updater fails due to missing appdata/firmware folder

jlewin, 2017-06-29:
**Expected**
MatterControl should create folder if missing

**Actual**
Directory is not created and downloaded firmware is not copied to disk at expected path

larsbrubaker, 2017-06-29:
Investigation: 
The firmware folder is only created in the user data directory if backup is enabled. Change it to always ensure the folder exists.
```
// Backup Firmware
await Task.Run(() =>
{
	// Create the firmware directory if needed
	Directory.CreateDirectory(FirmwareDirectory);

	RunAvrDude(GetFlashCommand(backupFirmwareLocation, TransferType.r));
});
```

jlewin, 2017-07-20:
@larsbrubaker  - Is this a design_tools bug? I thought this was observed on 1.7 Experimental builds when flashing Pulse machines

larsbrubaker, 2017-07-20:
This is happening in both. It is only when we disable the backup. At least
that is my recollection.