MatterHackers/MatterControl #2238: "Support Air Gap" prints above current layer

whosawhatsis, 2017-07-04:
As a test, I set the support air gap to a value that was much higher than I would actually use (about 2mm), in order to ascertain some things about how it functioned. I was surprised to find that it was now trying to print part of the support structure in air, way above the current layer.

[image]

This is, of course, with an absurdly large value for that setting, but I confirmed that it still prints above the current layer when using a reasonable value. If, for example, you're printing .1mm layers and you use an air gap of .3mm, a totally reasonable setting, that material will be extruded into open air above the layer, barely adhering, then the next several layers will have to plow through the out-of-place material. The expected behavior was for the support to stop lower, before reaching the part. Is this really how this was intended to work?

larsbrubaker, 2017-07-06:
This is the intended behavior. [Here](http://wiki.mattercontrol.com/SETTINGS/General/Support_Material/Support_Options/Air_Gap) is the documentation we have already written up. If this is not clear enough we can discuss more here and update our wiki.

Thanks.

Here is the raw url:
http://wiki.mattercontrol.com/SETTINGS/General/Support_Material/Support_Options/Air_Gap

AlbertoColucci, 2022-12-19:
The right behaviour should be stopping the support "air gap" millimiters/layers BEFORE the first part layer, not moving up by "air gap" millimiters for one single layer, then come back down. 

By doing the actual method, the first layer is shifter up, into nothing, then the printer comes back down. In this way the max real air gap is the layer thickness...

The slicer should build support up until  X - AIR_GAP, where X is the Z coordinate of the first part layer above the support