MatterHackers/MatterControl #2277: MatterControl ALWAYS crashes shortly after starting

GaelicGrime, 2017-07-19:
so ran exquisitely until a week before the official release of Debian Stretch, then started crashing EVERY TIME I ran it.

I assumed it was upgrade related, and as I had nothing pressing forgot about it until today when I do have something to print.  Same behavior. Same messages (included below). 

Completely removed all traces of MatterControl, started fresh, same result.  DO NOT USE the install script as downgrading mono is not an option, removes half of the software I use if I let the install script run.

Only difference noted is that after a complete removal and reinstall today, MatterControl did allow me to login, then crashed (ran that from the application launcher), before going back to crashing every time.

Debian stretch + KDE, some packages from Sid which /should not/ be related as far as I know mostly audio libraries so I can use kxstudio.

Do also have slic3r installed (as I did previously) from the official source.

Anything else I can provide information wise?

******* begin includes
_will_ on _deskwill_ [20170718.2106] ~ U_ MatterControl
Could not set X locale modifiers

** (mono:19244): WARNING **: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
Waiting for a connection on port 52711...

Unhandled Exception:
System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'System.Net.Sockets.Socket'.
  at System.Net.Sockets.Socket.ThrowIfDisposedAndClosed () [0x0001b] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.Socket.GetSocketOption (System.Net.Sockets.SocketOptionLevel optionLevel, System.Net.Sockets.SocketOptionName optionName) [0x00000] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at SuperSocket.ClientEngine.TcpClientSession.ProcessConnect (System.Net.Sockets.Socket socket, System.Object state, System.Net.Sockets.SocketAsyncEventArgs e) [0x00060] in <4ba236fbf29248a4985a92e6e3907e4a>:0 
  at SuperSocket.ClientEngine.AsyncTcpSession.SocketEventArgsCompleted (System.Object sender, System.Net.Sockets.SocketAsyncEventArgs e) [0x0000c] in <4ba236fbf29248a4985a92e6e3907e4a>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.OnCompleted (System.Net.Sockets.SocketAsyncEventArgs e) [0x0000e] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.FinishWrapperConnectSuccess (System.Net.Sockets.Socket connectSocket, System.Int32 bytesTransferred, System.Net.Sockets.SocketFlags flags) [0x00016] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.MultipleConnectAsync.Succeed () [0x0002d] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.MultipleConnectAsync.InternalConnectCallback (System.Object sender, System.Net.Sockets.SocketAsyncEventArgs args) [0x000b0] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.OnCompleted (System.Net.Sockets.SocketAsyncEventArgs e) [0x0000e] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.Complete () [0x00000] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.Socket+<>c.<.cctor>b__306_3 (System.IAsyncResult ares) [0x00057] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.SocketAsyncResult+<>c__DisplayClass27_0.<Complete>b__0 (System.Object _) [0x00000] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00008] in <a07d6bf484a54da2861691df910339b1>:0 
  at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00074] in <a07d6bf484a54da2861691df910339b1>:0 
  at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <a07d6bf484a54da2861691df910339b1>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'System.Net.Sockets.Socket'.
  at System.Net.Sockets.Socket.ThrowIfDisposedAndClosed () [0x0001b] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.Socket.GetSocketOption (System.Net.Sockets.SocketOptionLevel optionLevel, System.Net.Sockets.SocketOptionName optionName) [0x00000] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at SuperSocket.ClientEngine.TcpClientSession.ProcessConnect (System.Net.Sockets.Socket socket, System.Object state, System.Net.Sockets.SocketAsyncEventArgs e) [0x00060] in <4ba236fbf29248a4985a92e6e3907e4a>:0 
  at SuperSocket.ClientEngine.AsyncTcpSession.SocketEventArgsCompleted (System.Object sender, System.Net.Sockets.SocketAsyncEventArgs e) [0x0000c] in <4ba236fbf29248a4985a92e6e3907e4a>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.OnCompleted (System.Net.Sockets.SocketAsyncEventArgs e) [0x0000e] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.FinishWrapperConnectSuccess (System.Net.Sockets.Socket connectSocket, System.Int32 bytesTransferred, System.Net.Sockets.SocketFlags flags) [0x00016] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.MultipleConnectAsync.Succeed () [0x0002d] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.MultipleConnectAsync.InternalConnectCallback (System.Object sender, System.Net.Sockets.SocketAsyncEventArgs args) [0x000b0] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.OnCompleted (System.Net.Sockets.SocketAsyncEventArgs e) [0x0000e] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.SocketAsyncEventArgs.Complete () [0x00000] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.Socket+<>c.<.cctor>b__306_3 (System.IAsyncResult ares) [0x00057] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Net.Sockets.SocketAsyncResult+<>c__DisplayClass27_0.<Complete>b__0 (System.Object _) [0x00000] in <2b0d86369d72459baed0cee98a8e578a>:0 
  at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00008] in <a07d6bf484a54da2861691df910339b1>:0 
  at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00074] in <a07d6bf484a54da2861691df910339b1>:0 
  at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <a07d6bf484a54da2861691df910339b1>:0 
_will_ on _deskwill_ [20170718.2106] ~ U_ 

_will_ on _deskwill_ [20170718.2106] ~ U_ uname -a
Linux deskwill 4.9.0-3-amd64 #1 SMP Debian 4.9.30-2+deb9u2 (2017-06-26) x86_64 GNU/Linux
_will_ on _deskwill_ [20170718.2121] ~ U_

****** end include

GaelicGrime, 2017-07-19:
forgot:
Nvidia 375.66-1 (stable)
Mono 5.0.1.1-0xamarin5+debian8b1 (this came with the upgrade from Jessie to Stretch, Jessie stopped working with 4.6.2.7-dfsg-1 (Stable) prior to the dist-upgrade as mentioned in my first message.

I do not know if there was a decimal update in Nvidia drivers as part of the Jessie --> Stretch upgrade or just prior, will remember when I am at the offending machine to check.

nicolasdb, 2017-07-24:
Hi guys, 
Just to mention that I just had a similar problem problem with the soft crashing at start on Windows7 and mattercontrol 1.7.4 
probably a problem with the autoupdate but I tried to reinstall the last version manually, still crashing at start.
But this morning, the problem is solved now after full uninstall + cleaning (with revo uninstaller) and reinstall fresh download from mattercontrol website.

it's probably not related but... i don't know....

unlimitedbacon, 2017-07-24:
Thanks for the report This is a duplicate of https://github.com/MatterHackers/MatterControl/issues/1989. We're working on Linux builds now, so a release that fixes this issue should be out Soon™. In the mean time, there is a fix described in the other bug report.

closed 2017-07-24 by unlimitedbacon