MatterHackers/MatterControl #2291: Slice very slow

diver197, 2017-07-28:
Sliced hollow or the surface of the complex model, very slow, cura2.5 as long as less than 10 minutes, MC takes a few hours
1.7.4

jstefanop, 2017-08-04:
Agreed...need better hardware acceleration for slicing. Either use multithreading to do the slicing (this is a prefect application for multithreading), can probably even use GPU OpenCl based acceleration for even faster results.

larsbrubaker, 2017-08-04:
Can you please include the mesh that causes this error? Thank you.

DanielJoyce, 2017-08-10:
Gonna suggest the mesh may have defects like flipped normals or non manifold geometry

jstefanop, 2017-08-10:
Nope just upload any stl that's over 5mb or so. All other slicers can process large stls no problem, matterslice chokes past 1mb.

larsbrubaker, 2017-08-11:
Support for larger models is in progress. Thanks for the feedback.

colemorgan, 2018-11-12:
Posting to revisit this. I am printing small models for a board game and they each take 3-5 minutes vs Simplify3D 3-5 seconds. Is there anything I can do here?

blueeagle69, 2019-04-09:
Agreed.
I love MatterControl, it's my main slicer.
But the second supports are turned on the slicing time rockets.
The object i'm working on isn't that extreme.
But slicing with my beloved MatterControl took around 28 seconds.
Installed Cura 4 to test, and the slicing took 6 seconds.