MatterHackers/MatterControl #2342: MatterControl 1.7.4 always centered print

weirdguy500, 2017-08-11:
I'm using an Anet A8 with Skynet3D v 1.1
I have it connected as a Prusa Research i3 MK2
I have the center print on bed option unchecked, but whenever i do a print it starts off in the center
I would like to print elsewhere because my print bed is a bit uneven due to badly removing prints before. Will be getting a borosilicate glass plate in the future but would still like help with the center on bed issue

also. would anyone happen to know what i can do about re-positioning the center of my printer since i've replaced the stock extruder with a bowden extruder

larsbrubaker, 2017-08-11:
I am not able to reproduce this error. Can you please attach an export of your printer profile and the stl that is producing this error and we will look into what might be the problem?

Thank you.

weirdguy500, 2017-08-12:
here's the printer profile. and it's all STL files that are printing in center even with the print from center unchecked
[anet a8.zip](https://github.com/MatterHackers/MatterControl/files/1219352/anet.a8.zip)

tobor8man, 2017-08-12:
I have the same problem. However I notice that the center on bed option works with the matter slicer engine but not with the slic3r engine. Not sure how much that option is related to the slice engine. I've attached the the source STL and two small gcode files which are both constructed with the center on bed option off and the object approximately centered on position 160, 160. The slicer used is in the file name. I'm using Matter Control version 1.7.4 on Windows 7 with an Anet A8.

[center_on_bed_PASS_and_FAIL.zip](https://github.com/MatterHackers/MatterControl/files/1219522/center_on_bed_PASS_and_FAIL.zip)

larsbrubaker, 2017-08-21:
We will investigate this and fix it in a intermediate release. The bigger fix will be 2.0 release in which Center on Bed has been removed in favor of a much more interactive platting solution.

larsbrubaker, 2019-01-24:
Closing this as 2.0 is shipping and resolves this issue.

closed 2019-01-24 by larsbrubaker