MatterHackers/MatterControl #2439: Mac OS: Terminal hidden / enhancement idea

midnightcharger2, 2017-09-16:
Latest build of mattercontrol has the terminal hidden on macbook.  Cannot bring it to the front.  It does not even show in the dock.  I can see it running if I right-click on the Matter contol icon on the dock and "bring all windows to front" It ignores being clicked on and brings up the Mattercontrol GUI.  Restart of the macbook did not solve it.

This seemed to be a slight issue in previous builds also.  If the terminal was minimized, sometimes it was difficult to bring it to the front again.  Usually "bring all windows to front" would revive it.

I was wondering as an idea, could the Terminal be placed along with the "3D view / Layer View tabs to switch back and forth without going to the "file/view/help" menu section, since this is a common used item.

larsbrubaker, 2017-09-16:
Thanks for the feedback.

We have a new version coming out soon that moves the terminal much like you are suggesting. 

You can see the location for a second in this video
https://youtu.be/xnNKG5WhS60

closed 2017-09-16 by larsbrubaker

midnightcharger2, 2017-09-16:
Looks swanky!!  Thanks for the feedback.