MatterHackers/MatterControl #2442: Mattercontrol Ver 1.7.5

p1H3u7u7n5, 2017-09-16:
Mattercontrol keeps on interrupting the Active USB connection to my flashforge just as it starts a print it suddenly disconnects the USB. Need Help

larsbrubaker, 2017-09-19:
What does the terminal show? Does windows still show the USB port as available? It could be that the OS has lost connection, we should check that first.

p1H3u7u7n5, 2017-09-20:
Error is at end of Print Log

->Communication State: PreparingToPrint

->N61 M105*16

<-ok T0:26 B:19 T1:26
->N1 M110 N1*125

<-ok
->N1 M110 N1*125

<-ok
->N2 M114*37

->Communication State: Printing

<-ok C: X:0.0211640211640212 Y:0.0211640211640212 Z:0.005 E:0 [0.016]

; Generated with MatterSlice 1.0
; filamentDiameter = 1.75
; extrusionWidth = 0.4
; firstLayerExtrusionWidth = 0.4
; layerThickness = 0.2
; firstLayerThickness = 0.2
; automatic settings before start_gcode
->N3 G21*25 [0.017]

<-ok [0.017]

->N4 M107*33 [0.017]

<-ok [0.039]

->N5 M140 S60*86 [0.039]

<-ok [0.061]

->N6 M105*33 [0.968]

<-ok T0:26 B:19 T1:26 [1.032]

->N7 M105*32 [2.042]

<-ok T0:26 B:19 T1:25 [2.106]

->N8 M105*47 [3.115]

<-ok T0:27 B:19 T1:26 [3.179]

->N9 M105*46 [4.187]

<-ok T0:26 B:20 T1:26 [4.251]

->N10 M105*22 [5.260]

<-ok T0:26 B:21 T1:26 [5.324]

->N11 M105*23 [6.335]

<-ok T0:26 B:21 T1:26 [6.398]

->N12 M105*20 [7.409]

<-ok T0:27 B:23 T1:26 [7.473]

->N13 M105*21 [8.482]

<-ok T0:26 B:23 T1:26 [8.546]

[... 4024 lines cut ...]
<-ok [292.191]

->N1019 G1 X161.99 Y133.58 E21.099*98 [292.191]

<-ok [292.255]

->N1020 G1 X161.28 Y134.17 E21.13*92 [292.255]

<-ok [292.340]

->N1021 G1 X160.57 Y134.75 E21.16*85 [292.340]

<-ok [292.425]

->N1022 G1 X159.87 Y135.34 E21.191*107 [292.425]

<-ok [292.510]

->N1023 G1 X159.12 Y135.88 E21.221*105 [292.510]

<-ok [292.574]

->N1024 G1 X158.38 Y136.42 E21.252*102 [292.574]

<-o

p1H3u7u7n5, 2017-09-24:
I have not seen any recent techs on this issue I need this issue resolved
urgently as i cannot keep my FFC Woody to stay connected to the USB Please
advise i need answers now. I am not going to sit and wait

unlimitedbacon, 2017-09-25:
p1H3u7u7n5, it looks like your print log is incomplete. There is no error message at the end. Please upload the file using a service like [Pastebin](https://pastebin.com/) and post your response through GitHub, not through email.

Just a reminder, this is a free product. Making demands will not get you help faster.

p1H3u7u7n5, 2017-09-25:
->Communication State: PreparingToPrint

->N61 M105*16

<-ok T0:26 B:19 T1:26
->N1 M110 N1*125

<-ok
->N1 M110 N1*125

<-ok
->N2 M114*37

->Communication State: Printing

<-ok C: X:0.0211640211640212 Y:0.0211640211640212 Z:0.005 E:0 [0.016]

; Generated with MatterSlice 1.0
; filamentDiameter = 1.75
; extrusionWidth = 0.4
; firstLayerExtrusionWidth = 0.4
; layerThickness = 0.2
; firstLayerThickness = 0.2
; automatic settings before start_gcode
->N3 G21*25 [0.017]

<-ok [0.017]

->N4 M107*33 [0.017]

<-ok [0.039]

->N5 M140 S60*86 [0.039]

<-ok [0.061]

->N6 M105*33 [0.968]

<-ok T0:26 B:19 T1:26 [1.032]

->N7 M105*32 [2.042]

<-ok T0:26 B:19 T1:25 [2.106]

->N8 M105*47 [3.115]

<-ok T0:27 B:19 T1:26 [3.179]

->N9 M105*46 [4.187]

<-ok T0:26 B:20 T1:26 [4.251]

->N10 M105*22 [5.260]

<-ok T0:26 B:21 T1:26 [5.324]

->N11 M105*23 [6.335]

<-ok T0:26 B:21 T1:26 [6.398]

->N12 M105*20 [7.409]

<-ok T0:27 B:23 T1:26 [7.473]

->N13 M105*21 [8.482]

<-ok T0:26 B:23 T1:26 [8.546]

->N14 M105*18 [9.552]

[... 7740 lines cut ...]

->N1948 G1 X115.44 E45.171*2 [359.148]

<-ok [359.233]

->N1949 G1 X114.56 E45.2*4 [359.233]

<-ok [359.297]

->N1950 G1 X113.69 E45.229*12 [359.297]

<-ok [359.382]

->N1951 G1 X112.82 E45.258*15 [359.382]

<-ok [359.467]

->N1952 G1 Y146.87 F6600*26 [359.467]

Exception:A device attached to the system is not functioning.

Lost connection to printer.
->Communication State: FailedToConnect

->Read Thread Has Exited.

p1H3u7u7n5, 2017-09-28:
having to wait  days for an answer (this is poor Customer Service)

p1H3u7u7n5, 2017-10-01:
I need answers please FFC has been sitting unused due to your company lack
of CS and TS.

I will keep this up until i get answers

unlimitedbacon, 2017-10-02:
p1H3u7u7n5, If you are looking for customer service or technical support for a purchase you made from MatterHackers, then email support@matterhackers.com and include your order number in your message.

MatterControl is a free open source program, which you did not pay for. We are glad to answer questions and investigate bugs, but we are under no obligation to do so on any sort of timeline. 

GitHub is not a channel for customer support. It is a an organizational tool for software developers. If you are going to continue posting on GitHub, please review our [guidelines for contributing to MatterControl on GitHub](https://github.com/MatterHackers/MatterControl/blob/master/CONTRIBUTING.md).

closed 2017-11-17 by larsbrubaker