MatterHackers/MatterControl #2445: Validate Layer Height Description

3dstuffmakerusa, 2017-09-17:
The Help description which states "greater than nozzle diameter" should be changed to be "less than nozzle diameter"  to accurately reflect the function.

jlewin, 2017-10-30:
I'm not able to reproduce this issue. Neither the help text or the validation error shows the wrong logic. 

*Does this issue still occur and if so, on what versions?*

**Help Text**
[image]

**Validation Error**
[image]

jlewin, 2017-10-31:
Seems to be resolved in recent builds, please reopen in not

3dstuffmakerusa, 2017-10-31:
The setting is under Printer -> Features.  Select "Show Help" and read the  
comment for the Validate Layer height option.

jlewin, 2017-10-31:
Thanks for the clarification, I missed it in the config file as I was reviewing the layer_height field and the validate_layer_height field was not enabled. Should be resolved shortly.

closed 2017-11-27 by larsbrubaker