MatterHackers/MatterControl #2594: 1.7.5 X&Y Offset Error

kda406, 2017-10-26:
I just upgraded to 1.7.5 and now my parts print about 30mm right and 30mm to the rear of where I place them in the 3D View.  Prior versions printed on the print bed in the exact same location as I placed them in the 3D View.  

In fact, I printed something centered in the 3D View and it was centered on the bed with an older version (1.7.4?).  I popped the part off, upgraded the software, and every part since it has had an X and Y offset.

I'm using a FlashForge Creator X Dual clone with Sailfish firmware.

-Kyle

Profkanz, 2017-11-13:
I am having the same issue. Add part as AMF. Loads off center. Edit and move part to center, save. Switch to layer view, part is off center again.
[image],
[image],
[image]

whosawhatsis, 2017-11-14:
@Profkanz Actually, it IS printing centered, or at least its bounding box is centered. When printing on a circular platform, it's important to understand the difference between different types of centering. The software identifies the smallest rectangle aligned to the axes that the shape will fit into (known as the bounding box), then puts the center of that rectangle in the center of the platform. This works well for a rectangular platform, but for shapes that have features close to some corners of the bounding box and not others, that strategy can fail for a circular platform, as you're seeing here. The ideal solution requires using a bounding circle instead, but the smallest circle problem (https://en.wikipedia.org/wiki/Smallest-circle_problem) is harder to solve than simply finding the minimum and maximum coordinates in each dimension.

Profkanz, 2017-11-14:
Makes total sense since a similar part with four lobes does not have this problem.
dk

Profkanz, 2017-11-14:
I solved this problem by creating a six lobe base for the three lobe part. The base is .005 inch thick. I planned to use this like a raft or brim which would break away. The 3D view shows the thin base centered on the circular build plate. The slicer however apparently did not see the thin base and generated G-code to build the three lobe part. 
[image],
[image]

larsbrubaker, 2018-12-29:
looks like this is resolved

closed 2018-12-29 by larsbrubaker