MatterHackers/MatterControl #2601: Autolevelling on Rostock MAX v2

robotmaker, 2017-10-28:
I've installed a autolevelling probe on my Rostock Max V2 printer  using Rambo board and Repetier V091 firmware. It seems to works fine using G32 S02, G30,, G29 (wait disabled) - but that's just limited to 3 points which is not accurate enough. 

So I  configure MatterControl to use the probe during auto-leveling.  I want to use the probe in conjunction with the 13 point, 3x 3 mesh or other print levelling wizard. 

It seems to start OK and  - it probes the first point ....but then just sits and waits. The 'Next' button stay's grayed-out and there's no way to advance to the next probe position.

Have you got any ideas what to do? 

Thanks 
Colin

larsbrubaker, 2017-10-29:
The leveling with prob uses G30. The thing to do is look at the gcode output and see if something is going wrong there (that is obvious). If you don't see anything it would be great if you could save and zip the terminal output and add it to this bug (also please include your printer profile).

Thanks.

robotmaker, 2017-10-29:
Hi Lars,
The auto-levelling on the Rostock Max V2 starts by travelling down to the first point - in the same way as with the manual configuration. It goes up and down 2x on the first point; as setup in the setting/ Printer/Print levelling/Number of print samples. 
It finally  travels up to 5mm - but then just stops and  seems to be waiting for the next point.

I don't see anything wrong in the terminal with no additional gcode or error  messages. 
Attached is the printlog, Printer Profile and EPROM dump. 

For G32 S2 command I'm using the standard firmware from SeeMeCNC with the following changes in the Configurations.h:
Z_PROBE on pin 10 (which is the "ORIG_Z_MIN_PIN" on the Rambo board)  
#define FEATURE_AUTOLEVEL true
#define BED_LEVELING_METHOD 1
#define BED_CORRECTION_METHOD 0
//the following lines don't seem  to make any difference to the G32 command
#define BED_LEVELING_GRID_SIZE 5
#define BED_LEVELING_REPETITIONS 5
//
#define Z_PROBE_X1 0
#define Z_PROBE_Y1 -50
#define Z_PROBE_X2 -50
#define Z_PROBE_Y2 50
#define Z_PROBE_X3 50
#define Z_PROBE_Y3 50
#define Z_PROBE_X4 0
#define Z_PROBE_Y4 0

[SeeMeCNC Rostock MAX v2 (usb).txt](https://github.com/MatterHackers/MatterControl/files/1425260/SeeMeCNC.Rostock.MAX.v2.usb.txt)
.
[print_log.txt](https://github.com/MatterHackers/MatterControl/files/1425242/print_log.txt)

[eeprom_settings.txt](https://github.com/MatterHackers/MatterControl/files/1425249/eeprom_settings.txt)

Any ideas?

Colin

larsbrubaker, 2017-10-31:
It may be a bit before we can get to this, but at least you have given us some info to dig into.

Thank you.