MatterHackers/MatterControl #2619: Communication error with TCP to COM bridge

ligius-, 2017-11-02:
I'm using a NodeMCU (ESP8266, 3.3V) with a level converter to talk directly to a MightyBoard MCU (ATMega1280, 5.5V). The printer uses x3g/binary for serial communication.

If I connect a USB to serial converter directly on the pins it works just fine, I can control the printer via MatterControl and print without errors.
If I use a TCP to COM bridge instead (tried various programs and settings) the communication fails after a few commands with "Generic Packet Error, packet discarded".

To reproduce, I go into the control panel for the printer and nudge the X or Y steppers a bit in either direction. Eventually I get the above error in the terminal and communication ceases, the steppers are left powered (un-released).

I have not been able to figure out why, on my oscilloscope all the signals look ok. 
I'm suspecting it has to do with some latency or buffering, but I see no option to configure that.

Attaching a short snippet of communication that I was able to peek, the communication error starts around the middle part.

Also attaching oscilloscope traces (from another reproduction), the middle part shows communication loss. The blue trace is the TX (from computer), yellow is RX, though not all details can be captured.

[MC port error.txt](https://github.com/MatterHackers/MatterControl/files/1438539/MC.port.error.txt)

[image]

Here is both a serial dump (hex) and the terminal dump, they should be relatively in sync:

[print_log 2.txt](https://github.com/MatterHackers/MatterControl/files/1438593/print_log.2.txt)

[MC port error 2.txt](https://github.com/MatterHackers/MatterControl/files/1438594/MC.port.error.2.txt)

unlimitedbacon, 2017-11-02:
Couple of things.

1. We tried to do this on a Replicator 2 once. Although the Mightyboard has the pins for additional serial connections, the Sailfish firmware is not set up to talk on the auxiliary serial ports. It only uses the primary serial port, which goes directly to the USB/Serial converter. To use the other serial ports you would need to reconfigure the printer's firmware. We did not want to deal with that at the time.
2. I don't think MatterControl will speak S3G over a TCP connection. The S3G and TCP plugins are separate communications drivers. As far as I know, MatterControl can use one or the other, but cannot chain the two together. @mmoening, can you confirm this?
3. I've never been able to get a reliable connection using NodeMCU. The [esp-link](https://github.com/jeelabs/esp-link) firmware is much more robust.

mmoening, 2017-11-02:
I can confirm point 2.

I decided to wait on handling the chaining communication plugins until I had a case that required it. Furthermore due to the fact that replicators and the like required flashing in order to make an alternative serial input work @unlimitedbacon and I decided to recommend flashing marlin with the required changes.

ligius-, 2017-11-02:
Thank you for your prompt reply.

1. I used the standard MCU (UART0?) pins, the ones connected to the onboard usb-serial converter. Which was lucky, as that converted just burned out a week ago.

[image]

2. It's not a TCP connection but a standard serial one. The ESP8266 is working in serial bridge mode. As a matter of fact, using the exact same pins with a CH430 converter works just fine.

3. I'm using now esp-link but also tried it with https://github.com/beckdac/ESP8266-transparent-bridge and ESP3D, I have the exact same issue. I've used the name "NodeMCU" just as a board reference, not the firmware.

unlimitedbacon, 2017-11-02:
Ok I am confused now. Are you or are you not using the Networked Printing (TCP) feature in MatterControl?

If you are, then it's not going to work since MatterControl does not currently support Sailfish Communication (S3G) and Networked Printing (TCP) at the same time.

If you are not, then I don't think we can help you. Since it works over a normal USB/Serial connection, it's not a MatterControl problem.

ligius-, 2017-11-02:
Perhaps I should've been clearer. For this issue I am **not** using TCP, only 'standard' serial.
The other issue opened before this, was related to the fact that switching to TCP and back to serial breaks the printer profile.

It works over normal serial connection, but I think there are some hardcoded timings or some buffering/flow control that is not taken into account. I wish I could help more, but I don't know where to start.

Just to be clear again, this issue relates to MatterControl working over pc -> usb-> serial but not over pc -> virtual serial -> wifi -> serial.