MatterHackers/MatterControl #2760: [feedback] MC 1.7 win 10

mer2329, 2017-12-12:
hi this is not a bug just some feedback.

i just installed the 1.7 update and it is much slower then 1.6
maybe its the models im slicing but it just seems much, MUCH slower.
also has a lot of bottlenecks

so if i can find a 1.6 installer i will be rolling back

larsbrubaker, 2017-12-14:
Thank for the feedback. We are finishing up work on 2.0 and it is much faster in almost all ways. It should be out around March.

mer2329, 2018-01-01:
i just re read my post and i see a typo and i wanted to correct it.
i just installed the 1.7.5 update and it is much slower than 1.6

should have read 
i just installed the 1.7.5 update and it is much slower than 1.7.4

my post has been fixed