MatterHackers/MatterControl #2802: Setting extruder temperature under "Filament" does not change both extruders

stuartpb, 2017-12-24:
I've got a dual-extrusion PLA print going right now where I bumped the Extruder Temperature setting to 220C: the target for my *right* extruder's been set to 220C, but the *left* extruder is still targeting 205C.

larsbrubaker, 2017-12-26:
You need to set the second extruder Material settings to a material that has the temperature you want. This is confusing and will be replaced by settings per extruder in the future.

dcconn, 2018-02-21:
Any news on when this will be available? I.E. Having ability to enter settings for each extruder in "Settings->Filament->Temperatures"

larsbrubaker, 2018-03-08:
This is in the 2.0 build which will be available in alpha in the next few weeks.

You can also find it in our experimental build https://mattercontrol.appspot.com/downloads/matterhackers-experimental-windows/development

closed 2018-03-08 by larsbrubaker