MatterHackers/MatterControl #2884: Exporting printer settings

Banbury, 2018-01-13:
I have created custom settings for my printer. When exporting the profile, I have several problems:

* The baud rate is changed to  the default value of "250000" in OemLayer. The baud rate of my printer is fixed, so I don't want the default. 
* make and model are set to "Other", even though I gave proper values, when creating the profile.
* created_date is set to "7/19/2016 7:55:58 PM", even though I created the profile two days ago
* there should be an option to remove StagedUserSettings and UserLayer on export

larsbrubaker, 2018-01-20:
The settings you are wanting to bake into the OEM layer are usually done when the profile as added to our system. We take a user profile and our web services convert it into  an OEM profile.

As to exporting a 'backed' user profile, that is a feature that is not currently planned but is in our backlog of features.

Banbury, 2018-01-20:
Do you have a script, that does the baking? If you can release that, or at least describe, what it does, I could write something myself.