MatterHackers/MatterControl #2903: MatterControl crashes on Linux (Ubuntu 17.10) after adding STL to library.

jerrynsr, 2018-01-17:
Hi,
    I would like to use MatterControl with my Robo R1+ on linux.   There are issues when running on Ubuntu.  Whenever I add a new stl to the local library MatterControl segfaults:
*** Error in `mono': free(): invalid pointer: 0x000055d884e315c0 ***
Stacktrace:

  at <unknown> <0xffffffff>
  at (wrapper managed-to-native) GLib.SList.g_free (intptr) <0x0005f>
  at GLib.ListBase.Empty () <0x0013c>
  at GLib.ListBase.Dispose (bool) <0x0000f>
  at GLib.ListBase.Finalize () <0x0001d>
  at (wrapper runtime-invoke) object.runtime_invoke_virtual_void__this__ (object,intptr,intptr,intptr) <0x00068>

I am running the latest version of mono:
mono-4.0-gac/artful,artful,now 4.6.2.7+dfsg-1ubuntu1 all [installed,automatic]
mono-4.0-service/artful,artful,now 4.6.2.7+dfsg-1ubuntu1 all [installed,automatic]
mono-complete/artful,now 4.6.2.7+dfsg-1ubuntu1 amd64 [installed]
mono-csharp-shell/artful,artful,now 4.6.2.7+dfsg-1ubuntu1 all [installed,automatic]
mono-devel/artful,artful,now 4.6.2.7+dfsg-1ubuntu1 all [installed,automatic]
mono-gac/artful,artful,now 4.6.2.7+dfsg-1ubuntu1 all [installed,automatic]
mono-jay/artful,now 4.6.2.7+dfsg-1ubuntu1 amd64 [installed,automatic]
mono-mcs/artful,artful,now 4.6.2.7+dfsg-1ubuntu1 all [installed,automatic]
mono-profiler/artful,artful,now 4.2-2.1 all [installed]
mono-runtime/artful,now 4.6.2.7+dfsg-1ubuntu1 amd64 [installed,automatic]
mono-runtime-common/artful,now 4.6.2.7+dfsg-1ubuntu1 amd64 [installed,automatic]
mono-runtime-sgen/artful,now 4.6.2.7+dfsg-1ubuntu1 amd64 [installed,automatic]
mono-tools-devel/artful,artful,now 4.2-2.1 all [installed]
mono-utils/artful,now 4.6.2.7+dfsg-1ubuntu1 amd64 [installed,automatic]
mono-xbuild/artful,artful,now 4.6.2.7+dfsg-1ubuntu1 all [installed,automatic]
mono-xsp4/artful,artful,now 4.2-2.1 all [installed,automatic]
mono-xsp4-base/artful,artful,now 4.2-2.1 all [installed,automatic]
monodoc-base/artful,artful,now 4.6.2.7+dfsg-1ubuntu1 all [installed,automatic]
monodoc-gtk2.0-manual/artful,artful,now 2.12.40-2 all [installed,automatic]
monodoc-http/artful,artful,now 4.2-2.1 all [installed,automatic]
monodoc-manual/artful,artful,now 4.6.2.7+dfsg-1ubuntu1 all [installed,automatic]
ubuntu-mono/artful-updates,artful-updates,now 16.10+17.10.20171115-0ubuntu1 all [installed]

The work around currently is to restart MatterControl.   It's a PITA because I am working on a large project.   Lots of stl models.

UnixMonky, 2018-02-02:
I have the same issue, on Debian stretch. MC crashes whenever I add a .stl to the queue. I'm running MatterControl 1.7.5.9487.

Stacktrace:
  at <unknown> <0xffffffff>
  at (wrapper managed-to-native) GLib.SList.g_free (intptr) <0x0005f>
  at GLib.ListBase.Empty () <0x0013c>
  at GLib.ListBase.Dispose (bool) <0x0000f>
  at GLib.ListBase.Finalize () <0x0001d>
  at (wrapper runtime-invoke) object.runtime_invoke_virtual_void__this__ (object,intptr,intptr,intptr) <0x00068>

Native stacktrace:
	mono(+0xc8d94) [0x55fba1d33d94]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x110c0) [0x7fc11d3ad0c0]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcf) [0x7fc11ce18fcf]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x16a) [0x7fc11ce1a3fa]
	/lib/x86_64-linux-gnu/libc.so.6(+0x70bd0) [0x7fc11ce56bd0]
	/lib/x86_64-linux-gnu/libc.so.6(+0x76f96) [0x7fc11ce5cf96]
	/lib/x86_64-linux-gnu/libc.so.6(+0x777de) [0x7fc11ce5d7de]
	[0x41fb7670]

Debug info from gdb:
=================================================================
Got a SIGABRT while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

Aborted

mtfurlan, 2018-02-15:
I'm seeing the same thing also being on debian stretch with the same version of mattercontrol, 1.7.5.9487.
I'm getting a little bit more output for whatever reason
```
Time To Load Seconds: 0.10                                                                                                   
*** Error in `mono': free(): invalid pointer: 0x000055a9e90dab60 ***                                                         
======= Backtrace: =========                                                                                                 
/lib/x86_64-linux-gnu/libc.so.6(+0x70bcb)[0x7fa3b5207bcb]                                                                    
/lib/x86_64-linux-gnu/libc.so.6(+0x76f96)[0x7fa3b520df96]                                                                                    
/lib/x86_64-linux-gnu/libc.so.6(+0x777de)[0x7fa3b520e7de]                                                                      
[0x40fedea0]                                                                                                            
======= Memory map: ======== 
```
*memory map I'm excluding*
```
  at <unknown> <0xffffffff>                                                                                                                            
  at (wrapper managed-to-native) GLib.SList.g_free (intptr) <0x0005f>                                                                                  
  at GLib.ListBase.Empty () <0x0013c>                                                                                                                  
  at GLib.ListBase.Dispose (bool) <0x0000f>                                                                                                            
  at GLib.ListBase.Finalize () <0x0001d>                                                                                                     
  at (wrapper runtime-invoke) object.runtime_invoke_virtual_void__this__ (object,intptr,intptr,intptr) <0x00068>                             
                                                                                                                                             
Native stacktrace:                                                                                                                           
                                                                                                                         
        mono(+0xc8d94) [0x55a9e6023d94]                                                                                 
        /lib/x86_64-linux-gnu/libpthread.so.0(+0x110c0) [0x7fa3b575e0c0]                                                
        /lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcf) [0x7fa3b51c9fcf]                                                 
        /lib/x86_64-linux-gnu/libc.so.6(abort+0x16a) [0x7fa3b51cb3fa]                                                  
        /lib/x86_64-linux-gnu/libc.so.6(+0x70bd0) [0x7fa3b5207bd0]                                                    
        /lib/x86_64-linux-gnu/libc.so.6(+0x76f96) [0x7fa3b520df96]                                                       
        /lib/x86_64-linux-gnu/libc.so.6(+0x777de) [0x7fa3b520e7de]                                                       
        [0x40fedea0] 
```
Returns `134`.

The full output is here: https://gist.github.com/mtfurlan/a779c24de9aeceb19793d2f144c4af46

Interestingly, when I restart mattercontrol, the STL has been added and I can slice it and everything else seems to work.

retrotails, 2018-03-02:
commenting to say it happens to me as well, 1.7.5.9487 on a very fresh install of debian testing. it also seems to successfully add the .stl when I restart. worth noting it only happens with some models. when I export a model from blender, it needs scaled up 1000x. I tried scaling it before exporting and it no longer crashes mattercontrol.

SBKch, 2018-04-26:
1.7.5.9487 - same for me Ubuntu 17.10. It crashes but the file is added after starting the app again. It happened for every single .stl model I've added so far (using 'Add' on the bottom left corner to add directly to the printer queue).

bevinhex, 2019-01-08:
Same issue here. Tried to update mono to 5, but after encountering more issue, reverted to 4.6 version, which is included in debian by default.

unlimitedbacon, 2019-01-08:
We're trying to get a release of 2.0 for Linux out soon. Hopefully it should take care of this problem.