MatterHackers/MatterControl #2941: Mattercontrol.exe Seen as Malware

slcahoon, 2018-01-26:
Just passing this on from:
Craig T Myers
Craig T Industries
404-408-0734
craigtind@gmail.com
www.craigtindustries.com

"Thought I'd make you aware that the latest version of Bitdefender antivirus software will quarantine the mattercontrol.exe file calling it malware and prevent loading or reinstalling of the software. Have used it for nearly a year without issue prior to the recent update.

As a result of this I have lost all Mattercontrol files including data for my printers. It just disappeared! Never seen anything like it. Now in search of which folders to backup to prevent future data loss."

unlimitedbacon, 2018-02-01:
Looks like we can submit a false positive sample to Bitdefender here.
https://www.bitdefender.com/submit/

unlimitedbacon, 2018-02-05:
The version in question was 1.6.1. This is not an issue with 1.7.5. Nevertheless, the customer submitted a report to Bitdefender themselves.

closed 2018-02-05 by unlimitedbacon