MatterHackers/MatterControl #3093: Cannot log in on Linux mint.

michigansand, 2018-03-19:
I am trying to sync my printer settings from a win7 machine to Linux mint. When trying to login from the nux machine it just cycles and does not log in. I can log into the web site with no problems and on the win machine. No network connectivity issues on the nux machine except while logging in from the Mattercontrols software. Please help

unlimitedbacon, 2018-03-23:
What version of Mint are you on? What version of Mono do you have (run `mono --version`)? Can you run `mattercontrol` from the terminal and post the output after trying to log in?

michigansand, 2018-03-25:
HI Tyler,
I am running Mint 17.3, Mono 3.2.8. photo of out put listed below. Thank
you for your assistance in this.

Jeff Sand

unlimitedbacon, 2018-03-26:
@michigansand, Mint 17.3 is very old and includes an obsolete version of Mono. The latest version of Mono is 5.10. Follow the instructions here to upgrade it.

http://wiki.mattercontrol.com/Development/Running_on_Linux

Also in the future please respond through GitHub itself instead of through email. When you use email your attachments are not included.

michigansand, 2018-03-28:
This solved the problem. THANK YOU!

closed 2018-03-28 by unlimitedbacon