MatterHackers/MatterControl #3111: MatterControl Crashes when Clicking Generate in Layer View

JimMcKeever94, 2018-03-23:
I'm using an iMac running macOS Sierra. Matter Control Version is 1.7.1
Whenever I hit generate in the layer view MatterControl just shuts down without even giving me an error message. When I change the slicer engine from MatterControl to one of the other two options I either get a message saying a mesage saying Slicing Error, Please review your slice settings or it just does do anything but doesn't crash either. 

I've successfully installed and used MatterControl on two other computers but I'm not having this issue just on the the 3rd computer that I'm not using. The other two computers were a MacBook Pro and a Mac Pro. Both had no issues.
If anyone can help that would be greatly appreciated.

unlimitedbacon, 2018-03-23:
Can you gather a crash log? There are instructions here.
http://wiki.mattercontrol.com/Send_Debugging_Information#Crash_Logs

JimMcKeever94, 2018-03-24:
OK. Here's the what the terminal is outputting when I click generate in the Layer View.

Last login: Fri Mar 23 19:26:02 on console
Jamess-iMac:~ JamesMcKeever$ /Applications/MatterControl.app/Contents/MacOS/MatterControlMac ; exit;
objc[1538]: Class VCWeakObjectHolder is implemented in both /System/Library/PrivateFrameworks/AVConference.framework/Frameworks/ViceroyTrace.framework/Versions/A/ViceroyTrace (0xa8caf120) and /System/Library/PrivateFrameworks/AVConference.framework/Versions/A/AVConference (0xa8cca454). One of the two will be used. Which one is undefined.
/BuildRoot/Library/Caches/com.apple.xbs/Sources/AppleFSCompression/AppleFSCompression-96.30.2/Common/ChunkCompression.cpp:50: Error: unsupported compressor 8
/BuildRoot/Library/Caches/com.apple.xbs/Sources/AppleFSCompression/AppleFSCompression-96.30.2/Libraries/CompressData/CompressData.c:353: Error: Unknown compression scheme encountered for file '/System/Library/CoreServices/CoreTypes.bundle/Contents/Resources/Exceptions.plist'
/BuildRoot/Library/Caches/com.apple.xbs/Sources/AppleFSCompression/AppleFSCompression-96.30.2/Common/ChunkCompression.cpp:50: Error: unsupported compressor 8
/BuildRoot/Library/Caches/com.apple.xbs/Sources/AppleFSCompression/AppleFSCompression-96.30.2/Libraries/CompressData/CompressData.c:353: Error: Unknown compression scheme encountered for file '/System/Library/CoreServices/CoreTypes.bundle/Contents/Library/AppExceptions.bundle/Exceptions.plist'
Fontconfig error: Cannot load default config file

MatterSlice version 1.0

2018-03-23 21:37:13.723 MatterControlMac[1538:59791] 
Unhandled Exception:
System.FormatException: Input string was not in a correct format.
  at System.Number.ParseDouble (System.String value, NumberStyles options, System.Globalization.NumberFormatInfo numfmt) <0x90145a8 + 0x001a3> in <filename unknown>:0 
  at System.Double.Parse (System.String s, NumberStyles style, System.Globalization.NumberFormatInfo info) <0x9014578 + 0x0001f> in <filename unknown>:0 
  at System.Double.Parse (System.String s) <0x9014540 + 0x00023> in <filename unknown>:0 
  at MatterHackers.MatterSlice.ConfigSettings.SetSetting (System.String keyToSet, System.String valueToSetTo) <0xdc5f808 + 0x004c3> in <filename unknown>:0 
  at MatterHackers.MatterSlice.ConfigSettings.ReadSettings (System.String fileName) <0xdc5f4a8 + 0x0010f> in <filename unknown>:0 
  at MatterHackers.MatterSlice.MatterSlice.ProcessArgs (System.String[] args) <0xdc5aee8 + 0x00263> in <filename unknown>:0 
  at MatterHackers.MatterSlice.MatterSlice.ProcessArgs (System.String argsInString) <0xdc5a6a8 + 0x000f3> in <filename unknown>:0 
  at MatterHackers.MatterControl.SlicerConfiguration.SlicingQueue.CreateSlicedPartsThread () <0xdc54000 + 0x00f7f> in <filename unknown>:0 
  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) <0x7f8a6b8 + 0x00066> in <filename unknown>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, Boolean preserveSyncCtx) <0x769fd58 + 0x001c6> in <filename unknown>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, Boolean preserveSyncCtx) <0x769f558 + 0x00027> in <filename unknown>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) <0x7f8a608 + 0x0004f> in <filename unknown>:0 
  at System.Threading.ThreadHelper.ThreadStart () <0x7f8a4d0 + 0x0002b> in <filename unknown>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.FormatException: Input string was not in a correct format.
  at System.Number.ParseDouble (System.String value, NumberStyles options, System.Globalization.NumberFormatInfo numfmt) <0x90145a8 + 0x001a3> in <filename unknown>:0 
  at System.Double.Parse (System.String s, NumberStyles style, System.Globalization.NumberFormatInfo info) <0x9014578 + 0x0001f> in <filename unknown>:0 
  at System.Double.Parse (System.String s) <0x9014540 + 0x00023> in <filename unknown>:0 
  at MatterHackers.MatterSlice.ConfigSettings.SetSetting (System.String keyToSet, System.String valueToSetTo) <0xdc5f808 + 0x004c3> in <filename unknown>:0 
  at MatterHackers.MatterSlice.ConfigSettings.ReadSettings (System.String fileName) <0xdc5f4a8 + 0x0010f> in <filename unknown>:0 
  at MatterHackers.MatterSlice.MatterSlice.ProcessArgs (System.String[] args) <0xdc5aee8 + 0x00263> in <filename unknown>:0 
  at MatterHackers.MatterSlice.MatterSlice.ProcessArgs (System.String argsInString) <0xdc5a6a8 + 0x000f3> in <filename unknown>:0 
  at MatterHackers.MatterControl.SlicerConfiguration.SlicingQueue.CreateSlicedPartsThread () <0xdc54000 + 0x00f7f> in <filename unknown>:0 
  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) <0x7f8a6b8 + 0x00066> in <filename unknown>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, Boolean preserveSyncCtx) <0x769fd58 + 0x001c6> in <filename unknown>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, Boolean preserveSyncCtx) <0x769f558 + 0x00027> in <filename unknown>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) <0x7f8a608 + 0x0004f> in <filename unknown>:0 
  at System.Threading.ThreadHelper.ThreadStart () <0x7f8a4d0 + 0x0002b> in <filename unknown>:0 
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]

JimMcKeever94, 2018-03-26:
I just found something very interesting. So I'm currently trying to use MatterControl to slice prints directly to my new CR-10. However if i select the profile that I have for my Anet A8, Mattercontrol does not crash when I select generate in the Layer View. It slices the model fine. But when I switch bad to the CR-10 in the printer selection and I hit generate then, MatterControl crashes.

unlimitedbacon, 2018-03-26:
@larsbrubaker, can you look at this? The CR-10 profiles were created using 2.0. I'm guessing there is some backwards compatibility issue with using these on 1.7.

larsbrubaker, 2018-03-28:
Please rebuild these on 1.7 and make sure that avoids this problem.

We should not make any final 2.0 profiles until we ship. It is not really possible (certainly not worth the time) to go back and patch 1.7 to handle 2.0 profiles.

JimMcKeever94, 2018-03-28:
I appreciate you guys looking into this. I just wanted to let you know that I was able to figure out a work around for this by simply selected the manual configuration. This works pretty well but whenever I either unplug the printer or close MatterControl I have to redo the whole manual connection process every time I reopen MatterControl. 

-Jimmy

unlimitedbacon, 2018-03-29:
The problem is the Infill Overlap setting. 1.7 only accepts it in mm, not percentages. Changing it to 0.2 mm fixes the problem. I'll update the profiles.

larsbrubaker, 2018-03-30:
Thanks for the investigation

closed 2018-04-05 by larsbrubaker