MatterHackers/MatterControl #3238: can't move a folder in cloud library

larsbrubaker, 2018-04-25:
[image]

jlewin, 2018-05-09:
It's a little unclear looking at this how it might work. We're suggesting in this issue that we can move trees from arbitrary sources to arbitrary targets (given the suggestion it should work for local library). I had only ever imagined supporting this in CloudLibrary where top level move is supported as a simple operation. To accomplish this for other containers, we'd have to recursively walk the tree, creating a new container for each step, new child items for each step and purge the old content. This is incredibly complicated and error prone. 
- Cloud Library - top level move is supported, easiest scenario, likely would work without issue
- File System - top level move is supported but 
    - locking errors can and likely will abort mid-process destroying existing structure
    - no write operations are currently supported and user is directed to use Windows Explorer to accomplish tasks
- Local library - no top level move, must manually execute tree move logic for each container/sub container
- Unknown containers - ???

larsbrubaker, 2018-05-09:
I think at this time it only needs to work on cloud library. The example of local library is just a lack of understanding on my part.

closed 2018-05-16 by larsbrubaker