MatterHackers/MatterControl #3307: Plating History

blueeagle69, 2018-05-11:
With the Experimental build it appears you are unable to remove items from the Plating History.

larsbrubaker, 2018-05-14:
When you are in plating history selecting 'remove' will cause navigation to the folder with these files. That way you can quickly get to them and delete them in windows. 

Our default behavior with file system backed folders is to go to the operating system rather than delete from our app.

Thanks for the report it should be fixed in a build soon.

blueeagle69, 2018-05-15:
Hi.
I'm running Windows 10. It doesn't appear to be taking me to the location to where the folder is located. Could you possibly tell me the directory of the folder? I did have a look, but couldn't find it as yet.
But I could happily wait for it to be fixed in an upcoming build.
I just remembered too.
Email notifications is not working with the Experimental build.
Thanks.
Apologies, if this is already known.

blueeagle69, 2018-05-19:
Thanks for the fix. Windows now brings up the window to remove workspace files.
Unfortunately, now other weird bugs have crept in. Despite removing old workspace files, when next loading Matter Control, I get  random pyramid objects appearing in the history below the printer name. Clicking on them loads a blank workspace. Also, saving a workspace doesn't always work as intended. The next time I loaded Matter Control  I opened the workspace and ended up with multiple copies of the same object on the virtual bed! Six of them to be exact :)