MatterHackers/MatterControl #3366: Cannot open new empty bed (Experimental)

blueeagle69, 2018-05-30:
If I have had parts on the virtual bed, and printed etc.
If I then close that bed and attempt to open a new empty bed, or load another part from plating history all of a sudden, nothing will open.
Resulting in me having to quitting MatterControl and re-loading.
I can then open an empty bed, or existing object from plating history.

jlewin, 2018-06-01:
If I understand correctly, this is specifically an issue with closing Printer tabs.

We're tracking this problem on an internal bug and hope to have a fix soon. In the meantime the workarounds are to avoid closing your printer tab and you can restore the tab without closing MatterControl by changing theme colors, which will force a reload of the UI and restore a tab for the printer

MatterHackers/MCCentral#3400 - ActivePrinter not cleared on printer tab close

larsbrubaker, 2018-06-05:
Fixed in the next build

blueeagle69, 2018-06-10:
Hi.
You can now indeed change between tabs as long as there are objects loaded into the tab (bed) you open.
However you still cannot open a new empty bed.

jlewin, 2018-06-13:
Thanks again for the additional feedback. I can confirm what you're seeing and should have a fix in the next build

blueeagle69, 2018-06-13:
Thanks for the update 👍

blueeagle69, 2018-06-13:
I can confirm, this bug has now been fixed 👍 
Thanks for all your hard work. Especially today!

jlewin, 2018-06-13:
Great to hear and thanks as well for helping to improve the product

closed 2018-06-22 by larsbrubaker