MatterHackers/MatterControl #3403: Cannot Start a Print from Cloud Sync

blueeagle69, 2018-06-11:
Hi.

Whenever I attempt to start a print from Cloud Sync, Matter Control (Experimental) immediately closes.
(Windows 10)
Also, email notifications do not work. They do appear to work with the stable and beta builds though the last time I tried them.
Apologies, if any of these have been previously mentioned.

I have had my 3D printer a few months and Matter Control has become my one and only slicer.

Thanks.

larsbrubaker, 2018-06-12:
Thank you for the feedback. These issues should be addressed if you update to the latest build. Please re-submit this issue if you have problems after updating.

closed 2018-06-12 by larsbrubaker

blueeagle69, 2018-06-13:
Hi.

I have now updated to 2.x.9704.
I did a test, and started a print from Cloud Sync, and unfortunately MatterControl crashed again.
Have not had chance to test email notifications as yet.

blueeagle69, 2018-06-13:
(Update)
2.x.9706
Problem still persists.
Hope this helps.