MatterHackers/MatterControl #3428: current beta:  layer fan not triggering

bgoldman69, 2018-06-17:
current beta:  layer fan not triggering during print on pulse printer.
64 bit v 2.0.0.9713

larsbrubaker, 2018-06-22:
Can you please add the printer profile (or the model number) and the part you are printing to this issue in a zip?

Thank you.

jmosh11, 2018-07-23:
I am having the same issue with my lulzbot taz6 and the current beta.  I have to go into my printer's tune menu after the first 2 layers print and manually turn the cooling fans on.

I have tried a couple different parts and this has happened on all I have tried.

jmosh11, 2018-07-24:
FYI...I just tried something and it worked to make the cooling fans turn on...under the Fan settings there is the option "turn on if Below", I maxed the number out to 999 seconds and now my cooling fans turn on after the second layer is done.  

Seems like the software doesn't consider just one part but everything on the bed for the cooling fan so if you have a big part or a lot of small parts printing, the fan is not going to turn on.

closed 2018-12-05 by larsbrubaker