MatterHackers/MatterControl #3430: Failure to slice entire model - holes in gcode

MattHendricks, 2018-06-18:
I'm attempting to print something I have printed before. Last time all was great, sliced and printed just fine. I'm using build 2.0.0.9714, and it appears to be one of the components of the overall model is not being sliced.

[Model is available for download here](https://www.thingiverse.com/thing:2896212/files), you can ungroup in MatterControl and move the top and bottom sections around. The bottom section appears to have a triangle component that is not being sliced.

Thanks all, I have the old print at my desk if you want to see it.

jlewin, 2018-06-19:
[meshbase.zip](https://github.com/MatterHackers/MatterControl/files/2115488/E01E57E0E8D9AD63A7D2B0BA727BD2AD3D3DC7DB.zip)

The base portion of this mesh, specifically exhibits the issue. It appears to be at least two separate bodies 
 (per the backface/selection renderer) but ungroup fails to extract into discrete meshes:
[image]

It seems that rather than being a union of the bodies, it's acting more like a subtract:
[image]

jlewin, 2018-06-20:
Alec and Matt kept researching on this and confirmed that it was problems in the model

closed 2018-06-20 by jlewin