MatterHackers/MatterControl #3515: Top layers not slicing

gloddo, 2018-07-10:
I'm using MatterControl 2.0, and if I try to lift my model from the plate (to generate a full-body support), the top layers are not sliced.
If I lift 1mm from the plate, the model is truncated of 1mm on the top; if I lift by 2mm, it's truncated of 2mm and so on.

This happens with every model I try. 

What can it be? It's driving me crazy!

larsbrubaker, 2018-07-10:
Thanks for the report. We will investigate and fix.

jlewin, 2018-07-12:
Confirmed with reproducible case. Working on a fix, hopefully for the next release

jlewin, 2018-07-12:
Thanks again for this report. This issue has been resolved in MatterSlice and related items in MatterControl have been fixed as well. The next release will include the fix and feel free to report back if any issue remain.

-Cheers

closed 2018-07-19 by larsbrubaker