MatterHackers/MatterControl #3536: MatterControl 2.x for Linux as an AppImage

probonopd, 2018-07-16:
Would you be interested in packaging MatterControl 2.x for Linux as an Providing an [AppImage](http://appimage.org/)?

Doing so would have, among others, these advantages:
- Applications packaged as an AppImage can run on many distributions (including Ubuntu, Fedora, openSUSE, CentOS, elementaryOS, Linux Mint, and others)
- One app = one file = super simple for users: just download one AppImage file, [make it executable](http://discourse.appimage.org/t/how-to-make-an-appimage-executable/80), and run
- No unpacking or installation necessary
- No root needed
- No system libraries changed
- Works out of the box, no installation of runtimes needed
- Optional desktop integration with `appimaged`
- Optional binary delta updates, e.g., for continuous builds (only download the binary diff) using AppImageUpdate
- Can optionally GPG2-sign your AppImages (inside the file)
- Works on Live ISOs
- Can use the same AppImages when dual-booting multiple distributions
- Can be listed in the [AppImageHub](https://appimage.github.io/apps) central directory of available AppImages
- Can double as a self-extracting compressed archive with the `--appimage-extract` parameter

Especially in the 3D printing world, the format has gained a lot of traction, with most of the important slicers and CAD packages shipping official AppImages of release and pre-release versions:

- Ultimaker Cura
- Slic3r
- Slic3r Prusa Edition
- PrusaControl
- FreeCAD
- OpenSCAD
- QCAD

[Here is an overview](https://appimage.github.io/apps) of projects that are already distributing upstream-provided, official AppImages.

I would invest time into making this happen but since my Mono knowledge is rather limited I would do this only if MatterHackers is interesting in providing and supporting an official AppImage, and if someone from the MatterControl project would be willing to work with me along the way.

jlewin, 2018-07-16:
Thanks for the feedback and investigation. We'll look into this and report back

probonopd, 2018-07-20:
Work has started on https://gitlab.com/probono/MatterControlAppImage but I will likely need some support from the devs.

unlimitedbacon, 2018-07-20:
I think this is where the problem is happening.

https://github.com/MatterHackers/MatterControl/blob/e1836e5a5e07ba5c6288d0250b70f5fc3bb31c36/Program.cs#L46

MatterControl tries to find the location of it's executable and loads resources from the same directory. I don't know much about how AppImages work, but I'm guessing that while it's running inside of one it cannot get a valid path.

probonopd, 2018-07-20:
Excellent, that must be it. I am using `mkbundle` from Mono. Unfortunately I don't know how its inner workings are but seemingly it gets this part wrong. Can I turn on some debugging mode to see which path it is trying to access?

probonopd, 2018-07-20:
Are you saying you are changing the current working directory (as in `cd`)? That's generally not considered best practice. At least on Linux-type systems, you could derive the path to the application itself from `/proc/self/exe`, and construct a relative path to the resources from there.

probonopd, 2018-07-20:
This is what I found out from http://lists.dot.net/pipermail/mono-devel-list/2017-May/044329.html:

```
cat > assytest.cs <<\EOF
using System.Reflection;
using System.Text;
using Mono.Unix.Native;

public class Test
{
    public static void Main()
    {
        Assembly thisassy = Assembly.GetEntryAssembly();
        System.Console.WriteLine("Assembly Type: " + thisassy.GetType());
        System.Console.WriteLine(" CodeBase: " + thisassy.CodeBase);
        System.Console.WriteLine(" FullName: " + thisassy.FullName);
        System.Console.WriteLine(" Location: " + thisassy.Location);
        var builder = new StringBuilder(1024);
        Syscall.readlink("/proc/self/exe", builder);
        System.Console.WriteLine("BundleEnvironment: " +
builder.ToString());
    }
}
EOF

mcs -r /usr/lib/mono/4.5/Mono.Posix.dll assytest.cs

./assytest.exe 

Assembly Type: System.Reflection.MonoAssembly
 CodeBase: file:///home/me/assytest.exe
 FullName: assytest, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null
 Location: /home/me/assytest.exe
BundleEnvironment: /usr/bin/mono-sgen

/usr/lib/mono/4.5/mkbundle.exe  ./assytest.exe --deps -L  /usr/lib/mono/4.5/ -o assytest

./assytest
Assembly Type: System.Reflection.MonoAssembly
 CodeBase: file:///home/me/assytest.exe
 FullName: assytest, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null
 Location: assytest.exe
BundleEnvironment: /home/me/assytest
```

probonopd, 2018-07-20:
Give me some time, I am experimenting with it.

probonopd, 2018-07-20:
Running `/usr/lib/mono/4.5/mkbundle.exe MatterControl.exe --deps -L  /usr/lib/mono/4.5/ -L /usr/lib/mono/4.5/Facades/ -L /usr/lib/mono/4.5-api/ -o MatterControl` breaks the path finding. I haven't found the solution yet, partially because I have never ever worked with Mono. @unlimitedbacon can you have a look please?

unlimitedbacon, 2018-07-23:
I think you are on the right track. I know MatterControl loads a number of DLLs from the Release folder during startup. Sorry, I don't know enough about the startup process to help you further.

jlewin, 2018-07-23:
We are. You can find the code in PluginFinder.cs, LoadAssembliesFromFileSystem Line #20, where we load and iterate all assemblies in the bin folder looking for plugins. If I was to guess, it seems likely that the call to GetExecutingAssembly runs into the same issue that you resolved last week. I believe both cases were put in place to work around problems on Mac where the exe in the AppBundle is in a different physical location than the product assemblies.

jlewin, 2018-07-23:
You could try the following patch to see if it resolves the issue:
[patch.zip](https://github.com/MatterHackers/MatterControl/files/2221201/patch.zip)

jlewin, 2018-07-23:
I usually use git to apply. On my machine `git apply patch.diff` from root of the agg project works as expected.

jlewin, 2018-07-23:
Which would be `MatterControl/Submodules/agg-sharp`

probonopd, 2018-07-24:
Where can I see how you are creating the Mac .app bundle?

jlewin, 2018-07-24:
The Mac project would be even more confusing as it's essentially a wrapper around the normal solution which produces an exe with Xamarin.Mac bindings. 

Since your call stack suggests issues with libc/uname when running under the bundle, I'd try a less generic version of the OsInformationProvider to see if you can continue progressing forward. If it was me I'd replace the current WinformsInformationProvider with something like the following: 

```csharp
using Microsoft.VisualBasic.Devices;
using System;
using System.IO;

namespace MatterHackers.Agg.Platform
{
	public class WinformsInformationProvider : IOsInformationProvider
	{
		public WinformsInformationProvider()
		{
			this.OperatingSystem = OSType.X11;

			var size = System.Windows.Forms.Screen.PrimaryScreen.WorkingArea.Size;
			this.DesktopSize = new Point2D(size.Width, size.Height);
		}

		public OSType OperatingSystem { get; }

		public Point2D DesktopSize { get; }

		public long PhysicalMemory
		{
			get
			{
				var computerInfo = new ComputerInfo();
				return (long)computerInfo.TotalPhysicalMemory;
			}
		}
	}
}
```

jlewin, 2018-07-24:
Also, rather than apply modifications to the source as ad-hoc steps in the yml file, might it be easier to build off of a named branch that tracks progress of this task? If you were building off of said branch, I could have thrown in the above troubleshooting step in response to your feedback and just kicked off another build for evaluation, without as much manual effort on your end.

probonopd, 2018-07-24:
> Also, rather than apply modifications to the source as ad-hoc steps in the yml file, might it be easier to build off of a named branch that tracks progress of this task?

Please let me know the name of such a branch, then I can set it up accordingly. Thanks!

jlewin, 2018-07-25:
I forked your project and have been working through various issues trying to get to a point where I could reference something. I don't believe I'll have anything today but I'll likely have some additional questions tomorrow.

jlewin, 2018-07-25:
With a few caveats, I finally got a version of MatterControl to load via the generated AppImage. The icons aren't right and it wouldn't load until I put a local copy of the StaticData directory sitting next to the .AppImage file, but it's a start

You can see the artifact and revised .yml file at:
https://gitlab.com/lewin76/MatterControlAppImage/-/jobs/84127559

At first glance I believe whatever originally caused .Net to return the wrong paths on the GetExecutingAssembly calls is also causing .Net file system calls to escape the sandbox and get back to the normal filesystem. I'll keep investigating to make sure we're not doing anything funny, but baring that, is there anything special that might be required for .Net to see the packaged StaticData folder?

probonopd, 2018-07-25:
That's great progress. Thank you.

> is there anything special that might be required for .Net to see the packaged StaticData folder?

You need to load it from a location that you construct from the location of the running executable itself. For now I had just commented out the line that changes the current working directory because due to my lack of .net knowledge I did not really know what I was doing there. Pseudo-code: Instead of `open("./StaticData/some.thing");` do something like

```
location = Path.GetDirectoryName(System.Reflection.Assembly. GetExecutingAssembly().Location);
open(location + "StaticData/some.thing");
```

jlewin, 2018-07-25:
I can set about concatenating the current exe path with the relative paths previously used by the application but it feels like a step backwards. When we're run from any normal shell (Explorer, bash, Nautilus, Finder, etc..), the working directory is effectively the startup directory. The proposed workaround seems to be based on the premise that my application can no longer use relative paths from the working directory and must use absolute paths everywhere. That seems like a bit of a failure of AppImage to not match behavior that's consistent everywhere else and instead force a change in our application. Am I just thinking about this wrong?

jlewin, 2018-07-25:
This might actually be more of an mkbundle issue, still researching and trying to understand. 

My main concern is if I drop support for relative paths and convert to computed absolute paths, I just have to do it everywhere. There's probably only a couple of code places where we load content using relative paths but it's hard to know for sure and it's likely to break until they are all tracked down and updated.

jlewin, 2018-07-25:
Realizing this was all discussed above, definitely should have tried using .CodeBase as described above. Will work on some revisions and see how they perform

probonopd, 2018-07-25:
> I can set about concatenating the current exe path with the relative paths previously used by the application but it feels like a step backwards.

I think it is no problem to keep the change directory if you would like to continue using that logic, it just doesn't feel very clean to me but maybe it's a matter of personal preference. 

The only thing is that you need to change directory __not__ to `System.Reflection.Assembly.GetEntryAssembly().Location` but to `GetExecutingAssembly ` (I think), because the "entry assembly" points to the the wrong place. (Some experimentation may be needed.)

Alternatively we could have a small `AppRun` bash script inside the AppImage that can set and export paths, change to a certain directory etc. before launching the payload application. Let me know if you would use this route.

unlimitedbacon, 2018-07-25:
We already use a small launching script in the Debian package. This is `/usr/bin/mattercontrol`;

```
#!/bin/sh
cd /usr/lib/mattercontrol/
MONO_IOMAP=case exec mono /usr/lib/mattercontrol/MatterControl.exe $@
```

jlewin, 2018-07-25:
The reason for GetEntryAssembly versus GetExecutingAssembly I believe goes back to quirks on Mac where MatterControl is not the startup process and GetExecutingAssembly either had the wrong path or had null values in some contexts. At a glance I don't see details backing up that claim but I seem to recall working around problems in some platforms by switching away from GetExecutingAssembly. Continuing to test and troubleshoot...

probonopd, 2018-07-25:
We could replace the `AppRun` symlink with a simple `AppRun` file like this:

```
#!/bin/bash
HERE="$(dirname "$(readlink -f "${0}")")"
cd "${HERE}/usr/bin"
exec "${HERE}/usr/bin/MatterControl" "$@"
```

In this case we could leave the `Directory.SetCurrentDirectory` line commented out in the MatterControl source code.

jlewin, 2018-07-25:
The shell script may be needed as all attempts to detect the squash-fs style path continue to fail. I'd hope that with the shell script in place one of the Get*Assembly().CodePath values might be correct. 

I tried restoring the SetCurrentDirectory stuff, then disabled it and for a long while couldn't get anything to work. The problem finally revealed itself to be a small details I was skimming over, the .CodeBase path reports the filesystem rather than the AppImage/squashfs path. 

```
~/Desktop/asdf$ ./MatterControl-x86_64.AppImage
CodeBase: file:///home/matterhackers/Desktop/asdf/MatterControl.exe
...
Could not find a part of the path "/home/matterhackers/Desktop/asdf/StaticData/OEMSettings/Settings.json"
```

probonopd, 2018-07-26:
Changed https://gitlab.com/probono/MatterControlAppImage/blob/master/.gitlab-ci.yml to copy in the `AppRun` script.

probonopd, 2018-07-26:
I am beginning to ask myself whether `mkbundle.exe` generated ELF binaries even support loading .net assemblies like `.dll` files at all, since its goal is to make one executable that does not need external assemblies. Note that the `MatterSlicer` seems to work (it is not trying to load external `.dll` files apparently).

jlewin, 2018-07-26:
Thanks again for all the assistance. The added AppRun script works great. I merged in your changes, took out some troubleshooting code and we're up and running. With your update, the .Net `.CodeBase` properties now report usable values and my revisions to concatenate the startup and StaticData paths work as proposed

Assembly/Property | Path
---- | ----
Entry CodeBase |   file:///tmp/.mount_MatterQSrGlR/usr/bin/MatterControl.exe
ExecutingAssembly  CodeBase | file:///tmp/.mount_MatterQSrGlR/usr/bin/agg_platform_win32.dll
Entry Location |  MatterControl.exe
ExecutingAssembly  Location | agg_platform_win32.dll

A few quirks remain, which I'll continue looking into and just want to document while it's relevant:

- Launch failure if .Net is not installed  
  mono/mono#6187- mkbundle Linux - program doesn't run on system unless `mono-core` is installed  
  I tried a few attempts at setting config/machine-config but so far haven't had luck resolving the failure
- Need to update internal build scripts to participate with gitlab packaging or port to gitlab completely
- Need to package port workarounds for 250000 baud (`libSetSerial.so` - https://github.com/MatterHackers/MatterControl/blob/master/MatterControl.Printing/SerialHelper/build.sh)
- Need to ensure plugins get packaged and load as expected
- Research if dlls copied to `usr/bin` are redundant - expect `mkbundle --deps` to pull in what's required as part of the MatterControl bundle

probonopd, 2018-07-27:
> Launch failure if .Net is not installed

This, of course, is a major issue. The AppImage is supposed to run without the need of `mono-core` to be installed.

> mono/mono#6187- mkbundle Linux - program doesn't run on system unless mono-core is installed

Are you sure that this really the same issue? I suspect more that the plugins that are getting loaded from .dll files are not taken into account then `mkbundle.exe` calculates (and embeds) dependencies into the main binary.

Maybe we need to instruct it to process all *.dll files in the directory like mentioned in https://stackoverflow.com/a/27536070?

me@host:~$ wget https://gitlab.com/lewin76/MatterControlAppImage/-/jobs/84690721/artifacts/raw/MatterControl/bin/Release/MatterControl-x86_64.AppImage
me@host:~$ chmod +x MatterControl-x86_64.AppImage 
me@host:~$ ./MatterControl-x86_64.AppImage 
System.TypeInitializationException trying to get all screens: The type initializer for 'System.Windows.Forms.XplatUI' threw an exception.

Unhandled Exception:
System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.TypeInitializationException: The type initializer for 'System.Windows.Forms.Screen' threw an exception. ---> System.TypeInitializationException: The type initializer for 'System.Windows.Forms.XplatUI' threw an exception. ---> System.TypeInitializationException: The type initializer for 'System.Drawing.GDIPlus' threw an exception. ---> System.DllNotFoundException: libgdiplus.so.0
  at (wrapper managed-to-native) System.Drawing.GDIPlus.GdiplusStartup(ulong&,System.Drawing.GdiplusStartupInput&,System.Drawing.GdiplusStartupOutput&)
  at System.Drawing.GDIPlus..cctor () [0x000b0] in <b8b6ffaf51344842a77744e1fb2e4390>:0 
   --- End of inner exception stack trace ---
  at (wrapper managed-to-native) System.Object.__icall_wrapper_mono_generic_class_init(intptr)
  at System.Drawing.Graphics.FromHdcInternal (System.IntPtr hdc) [0x00000] in <b8b6ffaf51344842a77744e1fb2e4390>:0 
  at System.Windows.Forms.XplatUIX11.SetDisplay (System.IntPtr display_handle) [0x00073] in <69b1a555258c430e86ee1f3ef27d8ea6>:0 
  at System.Windows.Forms.XplatUIX11..ctor () [0x00077] in <69b1a555258c430e86ee1f3ef27d8ea6>:0 
  at System.Windows.Forms.XplatUIX11.GetInstance () [0x00019] in <69b1a555258c430e86ee1f3ef27d8ea6>:0 
  at System.Windows.Forms.XplatUI..cctor () [0x00066] in <69b1a555258c430e86ee1f3ef27d8ea6>:0 
   --- End of inner exception stack trace ---
  at System.Windows.Forms.Screen..cctor () [0x00034] in <69b1a555258c430e86ee1f3ef27d8ea6>:0 
   --- End of inner exception stack trace ---
  at MatterHackers.Agg.Platform.WinformsInformationProvider..ctor () [0x0000d] in <5ca4ce1190ec4f8eab82b26aa14f1bc2>:0 
  at (wrapper managed-to-native) System.Reflection.MonoCMethod.InternalInvoke(System.Reflection.MonoCMethod,object,object[],System.Exception&)
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) [0x00002] in <b0e1ad7573a24fd5a9f2af9595e677e7>:0 
   --- End of inner exception stack trace ---
  at System.Reflection.MonoCMethod.InternalInvoke (System.Object obj, System.Object[] parameters) [0x00014] in <b0e1ad7573a24fd5a9f2af9595e677e7>:0 
  at System.RuntimeType.CreateInstanceMono (System.Boolean nonPublic) [0x000a8] in <b0e1ad7573a24fd5a9f2af9595e677e7>:0 
  at System.RuntimeType.CreateInstanceSlow (System.Boolean publicOnly, System.Boolean skipCheckThis, System.Boolean fillCache, System.Threading.StackCrawlMark& stackMark) [0x00009] in <b0e1ad7573a24fd5a9f2af9595e677e7>:0 
  at System.RuntimeType.CreateInstanceDefaultCtor (System.Boolean publicOnly, System.Boolean skipCheckThis, System.Boolean fillCache, System.Threading.StackCrawlMark& stackMark) [0x00027] in <b0e1ad7573a24fd5a9f2af9595e677e7>:0 
  at System.Activator.CreateInstance (System.Type type, System.Boolean nonPublic) [0x00020] in <b0e1ad7573a24fd5a9f2af9595e677e7>:0 
  at System.Activator.CreateInstance (System.Type type) [0x00000] in <b0e1ad7573a24fd5a9f2af9595e677e7>:0 
  at MatterHackers.Agg.Platform.AggContext.CreateInstanceFrom[T] (System.String typeString) [0x00010] in <e83e5811429c49dbbb5c700ae287cbd7>:0 
  at MatterHackers.Agg.Platform.AggContext.get_OsInformation () [0x00016] in <e83e5811429c49dbbb5c700ae287cbd7>:0 
  at MatterHackers.Agg.Platform.AggContext.get_OperatingSystem () [0x00000] in <e83e5811429c49dbbb5c700ae287cbd7>:0 
  at MatterHackers.MatterControl.Program.Main () [0x00032] in <9ef3c78f48084576b544ec75c3c1e46d>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.TypeInitializationException: The type initializer for 'System.Windows.Forms.Screen' threw an exception. ---> System.TypeInitializationException: The type initializer for 'System.Windows.Forms.XplatUI' threw an exception. ---> System.TypeInitializationException: The type initializer for 'System.Drawing.GDIPlus' threw an exception. ---> System.DllNotFoundException: libgdiplus.so.0
  at (wrapper managed-to-native) System.Drawing.GDIPlus.GdiplusStartup(ulong&,System.Drawing.GdiplusStartupInput&,System.Drawing.GdiplusStartupOutput&)
  at System.Drawing.GDIPlus..cctor () [0x000b0] in <b8b6ffaf51344842a77744e1fb2e4390>:0 
   --- End of inner exception stack trace ---
  at (wrapper managed-to-native) System.Object.__icall_wrapper_mono_generic_class_init(intptr)
[... cut ...]
  at System.Activator.CreateInstance (System.Type type) [0x00000] in <b0e1ad7573a24fd5a9f2af9595e677e7>:0 
  at MatterHackers.Agg.Platform.AggContext.CreateInstanceFrom[T] (System.String typeString) [0x00010] in <e83e5811429c49dbbb5c700ae287cbd7>:0 
  at MatterHackers.Agg.Platform.AggContext.get_OsInformation () [0x00016] in <e83e5811429c49dbbb5c700ae287cbd7>:0 
  at MatterHackers.Agg.Platform.AggContext.get_OperatingSystem () [0x00000] in <e83e5811429c49dbbb5c700ae287cbd7>:0 
  at MatterHackers.MatterControl.Program.Main () [0x00032] in <9ef3c78f48084576b544ec75c3c1e46d>:0

probonopd, 2018-07-27:
> Need to update internal build scripts to participate with gitlab packaging or port to gitlab completely

> I just used GitLab as a matter of personal preference; it should be possible to achieve the same result by using Travis CI with GitHub. Let me know if I should do the conversion.

> Need to package port workarounds for 250000 baud (libSetSerial.so - https://github.com/MatterHackers/MatterControl/blob/master/MatterControl.Printing/SerialHelper/build.sh)

It should be easy to bundle a private `.so` library.

> Research if dlls copied to usr/bin are redundant - expect mkbundle --deps to pull in what's required as part of the MatterControl bundle

I assumed those are plugins loaded by the application at runtime, am I wrong? If no, they may be not needed after all...

probonopd, 2018-07-27:
Maybe we are now at a point where we need someone with intimate Mono knowledge like @directhex to look into why `mkbundle.exe` is failing us here...

jlewin, 2018-07-27:
>Launch failure if .Net is not installed

This looks like a straight-forward occurrence of mono/mono#6187 - same  complications with libc as detailed in that issue. I just haven't had any time yet to sort out how to pass the correct mkbundle parameters for this case and it didn't work the first few attempts I made.

As far as plugins go, they likely will need to be considered by mkbundle to package their dependencies but that's a later stage issue, once some actually get added to the packaging process.

@directhex - in mono/mono#6187 you guys discuss this problem and appear to suggest the solution is to pass `--config` and `--machine-config` params to mkbundle to resolve. I'll try the suggestion though, thx

jlewin, 2018-07-27:
@probonopd - no need to switch to Travis; using GitLab has been a great experience. We currently use AppVeyor, Travis with GitHub but I'm impressed with GitLab and appreciate working with and learning a new set of tools

probonopd, 2018-07-27:
Getting segfaults on 790accc0 and 2fd59d2b. Not on 7f61806a but there `libgdiplus.so.0` is missing. Perhaps we need to bundle this .so and its "unusal" dependencies in the AppImage?

jlewin, 2018-07-27:
My takeaway from the 6187 issue was binding errors that appear on systems lacking Mono can initially be attributed to incorrect dllmaps, which should be mitigated with the `--config` and `--machine-config` mkbundle params. 

I tried this, it didn't work but I'm hopeful that I simply did it wrong and/or the options conflict with other params being passed. I think we may also need to cross compile as well but for a Windows developer like myself, this all takes time to work through and I'm attempting to mix this task in with my existing workload, so it's slow going.  Hopefully with a few more days we'll have a build that works on machines that lack Mono, for now I'm elated with the progress that's been made and the tasks that remain

johnmark, 2019-04-21:
Is there any more progress on this? Would love to have a Linux package.

miguipda1, 2019-04-30:
Dear probonopd, 
did you progress on this project because I also need it to be able to use it on my Linux workstation.
Thank you.

probonopd, 2019-05-01:
Hi @miguipda1 unfortunately I did not fully succeed yet to make an AppImage out of this application, partly because I don't really know anything about how Mono works. I guess I will need help from someone who knows Mono. I am still happy to help (I know AppImage really well).

jimmy00784, 2019-05-07:
Hi @probonopd. I am familiar with Mono and Linux but not as much with AppImage creation. I've built MC 2.x beta on Linux in the past and even had to find my way around a timing issue happing inside AggSharp. Perhaps we can collaborate on this. Please let me know if I can be of any help.

probonopd, 2019-05-07:
Hi @jimmy00784 thanks for offering your help. I'd be more than happy to work with you on this. To get started, can you have a look at https://gitlab.com/lewin76/MatterControlAppImage/-/jobs/84690721/artifacts/raw/MatterControl/bin/Release/MatterControl-x86_64.AppImage and see what is going wrong? You can inspect the contents of the AppImage by running it with the `--appimage-extract` parameter.

jimmy00784, 2019-05-07:
@probonopd Initial findings:
Running AppImage: App loads but UI remains in a somewhat unresponsive state. The button hover effect is clearly visible but won't respond to clicks.
Running the extracted application: App loads and seems to be functioning without any errors.
The noticeable difference seems to be at startup in console messages.
When running AppImage, the console output stops at 
`Time to '': 209`
When running the extracted binary, the console output shows all of the below:
```
Time to '': 583
Time to 'ShowAsSystemWindow': 270
Time to 'First draw->RunOnIdle': 0
Time to 'MatterControlApplication.Initialize': 0
Time to 'PlatformInit': 0
Time to 'ApplicationController': 200
Time to 'ProfileManager': 29
Time to 'MainView': 0
Time to 'Plugins': 15
Time to 'Process Commandline': 0
Time to 'OnLoadActions': 7
Time to 'AddChild->MainView': 448
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: cHRM chunk does not match sRGB
Time To Load Seconds: 2.49
```

The individual "Time to" messages are printed by MatterControl modules that are loaded at startup.
The absence of those when running AppImage tells me that the AppImage run is running into a deadlock of some sort resulting in failure to load all necessary modules.

jimmy00784, 2019-05-07:
@probonopd with strace, I get a similar unresponsive UI. There might be a timing issue here. I'll try and rebuild the code on my machine and see if I can replicate the issue.

probonopd, 2019-05-07:
You can also run the tests on the extracted contents of the AppImage, although I doubt it'll make a big difference.

jimmy00784, 2019-05-07:
If I recall correctly, a few months ago, I ran into a similar issue with a local build where the UI wouldn't respond in a similar fashion. It ended up being a small timeout value on the UI refresh. I had a bug report open for that issue as well. I'll rebuild again to see if that issue is still there.
Where exactly are you stuck with the AppImage?

probonopd, 2019-05-07:
> Where exactly are you stuck with the AppImage?

https://github.com/MatterHackers/MatterControl/issues/3536#issuecomment-408471063

jimmy00784, 2019-05-07:
Local build runs without any issues.
How can I build an AppImage from it?

probonopd, 2019-05-07:
> How can I build an AppImage from it?

Follow each step from https://gitlab.com/lewin76/MatterControlAppImage/blob/master/.gitlab-ci.yml

probonopd, 2020-08-16:
Just stumbled across this [.NET Core AppImage example](https://github.com/ppy/osu-deploy/blob/697a49e9602502a2b7a899c0dff5383f6512d5d2/Program.cs#L207-L243) of how to deploy .NET Core (Mono) applications as an AppImage using `dotnet publish -f netcoreapp3.1 -r linux-x64` from within a `.cs` program.

Maybe it can be useful for MatterControl @jlewin?

jlewin, 2020-08-25:
@probonopd  - thanks for all your help on this issue and for continuing to staying interested in the solution. At first glace I'd think we can't compile to the netcoreapp3.1 framework/profile as we require Mono on Linux for Winforms support. Microsoft announced netcore Winforms support but it's only on Windows and they have yet to announce cross platform support. Further, Mono never kept up the Winforms effort and we've been plagued by Linux crashes due to poor support. It's plausible that the Winforms layer could be replace with GTK+ or similar however work on that has remained on the back burner for a long time.

I'm no longer with MatterHackers but I remain interested in this option and may resume looking at it in the future and will definitely consider the netcore AppImage example if that happens, thanks.

probonopd, 2020-08-25:
Thanks for the insights @jlewin - I don't even pretend to know all the subtle differences between the different .NET and Mono implementations from Ximian, Xamarin, Microsoft, and whatnot. But now that Microsoft itself owns Mono maybe the two will one day be merged altogether.

jlewin, 2020-08-25:
>But now that Microsoft itself owns Mono maybe the two will one day be merged altogether

Definitely, I believe that's already in the works and is what .NET 5 will be. Unfortunately Microsoft has only committed to .NET 5  Winforms support on Windows and from what I've gathered the prototype in Mono won't be in the .NET 5 Linux package. Maybe I have it wrong though. 

We only use Winforms as the OpenTK implementation seemed to provide decent cross platform support. agg-sharp can easily be revised to run on alternative windowing systems and hopefully a .NET 5 compatible one will emerge and this roadblock will be resolved. Ultimately Microsoft needs a cross platform 3D API for .NET and I remain hopeful that they secretly have support planned and coming in a near future release. So far though, that remains a fantasy.

nerdCopter, 2021-02-28:
+1 for AppImage. hopefully avoiding installation of mono-complete, but fine either way.

Marieholm20, 2022-12-28:
Any progress for Linux Mint 21 and matter control 2. for Linux support using appimage ?
It should be fantastic !
Lars.herrnsdorf@gmail.com