MatterHackers/MatterControl #3552: File reader

Profkanz, 2018-07-19:
New version does not add AMF file correctly.

larsbrubaker, 2018-07-19:
Thank you for reporting this error.

If possible:
- Can you add the problem file in a zip? 
- Can you please describe exactly what steps to reproduce

Profkanz, 2018-07-19:
Steps:
Right click STARTDblClick  LIBRARYDblClick LOCAL LIBRARYClick ":" (open menu)Click ADDFind file on PC and Click FILENAME.AMFClick OPEN
A folder opens in LOCAL LIBRARY with FILENAME GRAPHIC is this 

File is an AMF file output by Solidworks. File works fine in MC 1.7

larsbrubaker, 2018-07-19:
Would you be able to attach an AMF file that is failing in a zip to this bug?

I have many AMFs that are working if you have or can create one that doesn't I'll add it to our test suit.

Profkanz, 2018-07-19:
I have attached the AMF file at issue here. Note that the filename is the same as one included in the library. I have renamed the original file to avoid a filename conflict.

Profkanz, 2018-07-19:
PS.Same file as an STL file loads OK.
Also;I have notice that in previous versions STL files created from an inch dimensioned Solidworks part had to be scaled  to bring them up to the correct size. In MC2 these same files load as already upscaled objects. Is this a new feature by design or is something else going on?

Profkanz, 2018-07-21:
Update,
I too have other AMF files. All out of Solidworks 16.All the others load to MC2 Local Library without any problem.
So I opened the PART file in Solidworks, checked all the sketches to ensure that all are fully qualified, rebuilt the PART file and saved, deleted the AMF file, then again saved as AMF file.Add file to MC2 Local Library, same problem occurs.
Very strange.

krolco, 2018-08-06:
@Profkanz It doesn't look like your file actually got attached to the issue? I think you need to attach it on Github, instead of attaching it to an email reply.

Profkanz, 2018-08-06:
Just sent two zip files on Github.

krolco, 2018-08-06:
Thanks. I'm able to reproduce the behavior with the Mousetrap model. Hopefully someone will take a look soon (possibly me).

krolco, 2018-08-06:
OK, a bit more info for whoever picks this up (again, possibly me).
There is a NullReferenceException thrown at MeshContentProvider.cs line 77
Object3D.Load returns null here, and the null isn't checked at line 77 when the path is assigned. There is a "hasValidMesh: 0" in the output window, so clearly the loader doesn't like something about the AMF. More to investigate.

krolco, 2018-08-06:
More info: AmfDocument.cs line 436 or so. The case for v1 is hit on every face, but v2 and v3 are never hit, so every face is rejected. That leaves a mesh with vertices, but no faces, and everything fails downstream after that. Still investingating why v2 and v3 don't get read out of this particular AMF.

krolco, 2018-08-08:
OK, I have a fix for this. Making a PR.

krolco, 2018-08-08:
https://github.com/MatterHackers/agg-sharp/pull/1002

jlewin, 2018-08-08:
Thanks for everyone's assistance on this. We've been eagerly awaiting the repro files but didn't expect to get a fix as well. Very much appreciated! 

We'll write up a test for the missing whitespace case so we don't regress in the future and get this merged asap.

Thank again!

Profkanz, 2018-08-17:
All ok now.
Thanks all.

closed 2018-08-17 by Profkanz