MatterHackers/MatterControl #3559: Please provide a desktop file for Linux

probonopd, 2018-07-20:
Please provide a desktop file for Linux.

probonopd, 2018-07-20:
Something like this?

```
[Desktop Entry]
Name=MatterControl
Comment=Control your 3D printer
Exec=MatterControl
Icon=mattercontrol
Type=Application
Categories=Development;Engineering;
```

probonopd, 2018-07-20:
Also, is ther ean icon file besides https://github.com/MatterHackers/MatterControl/blob/master/application.ico in png format that can be used for Linux?

probonopd, 2018-07-20:
This is needed for https://github.com/MatterHackers/MatterControl/issues/3536 and https://gitlab.com/probono/MatterControlAppImage.

unlimitedbacon, 2018-07-20:
You can find all of these things in the [MatterControl AUR package](https://aur.archlinux.org/packages/mattercontrol/) for Arch or by unpacking the [official Debian package](https://mattercontrol.appspot.com/downloads/mattercontrol-linux/release). I suppose it would be helpful if they were added to the main MatterControl repository as well.