MatterHackers/MatterControl #3582: Compiling and DesignTools (EditorTools.dll)

dexmas, 2018-07-31:
Perfect idea of this tool, whant to add some smal features. I've compile it on VS2013 without any problem, but I have a question: in compiled version is epsion design tools (no primitives for design in left bar). But when I put EditorTools.dll from officialy installed MatterControl 2.0 - they apears. It seems I've find design primitives classes in project (DesignTools sources folder), but as I can see - they not compiles into any .dll as plugin and not used anywhere in .exe. So is any part of project missed on github, or I did smth. wrong?) Thanks.

larsbrubaker, 2018-08-01:
Some of our tools (particularly web components) are in separate private repositories. Sometimes we also develop new features here. All of design tools started out as private and we are moving more and more of it to our open repositories. The primitives are one of the components that we are planning to move into the open source part of our project.

Thanks for the feedback and interest.

dexmas, 2018-08-01:
Thank you for your answer. Still watching this cool project development.

closed 2018-08-01 by dexmas